gpiozero/gpiozero #2: License?

waveform80, 2015-09-15:
Just putting together a PR to start work on the docs and I was wondering what license you want to bung this under? (it's necessary for part of the Sphinx setup basically)

bennuttall, 2015-09-15:
Yeah I didn't do any of the packaging work until I was sure it was going to go ahead. Will add it all today.

waveform80, 2015-09-15:
If you want I can do the packaging bits in the docs PR - basically my usual Sphinx setup pulls all the package metadata from the setup.py (best to keep everything in one place). I've got a rather nice one that compensates for a variety of issues (Sphinx under Python 3.2 has *really* annoying deps due to various packages being 3.3+ compat only except for older versions). About the only thing I need for the packaging though is what license you want to bung this under.

bennuttall, 2015-09-15:
Added licence in a8b93cf97f9e5923594beda4117941d570e5bfe6

closed 2015-09-15 by bennuttall