gpiozero/gpiozero #3: Why is pin optional?

waveform80, 2015-09-15:
If *pin* is a mandatory argument to the constructor `InputDevice` why not simply make it a mandatory argument rather than checking for `None` in the constructor? Or is this an attempt to provide the friendliest possible error message in the event that a user forgets the one mandatory argument (i.e. instead of the usual "expected 2 arguments instead of 1" stuff)?

bennuttall, 2015-09-15:
Exactly that. There's scope to improve the error message in the exception I raise - but that's the idea.

waveform80, 2015-09-15:
Hmmm ... good intent, but that does mean any descendent class that wishes to introduce a mandatory constructor parameter is now forced to make it an optional parameter and code accordingly. Must admit I can't think of a case off the top of my head, but my gut instinct is that while this makes the error message a little simpler, the existing one isn't *that* bad and it is making another (admittedly advanced) use case more complicated.

Then again, the whole point of the library is to make the simplest possible interface for kids to use. So yeah, I'm persuaded.

closed 2015-09-15 by waveform80