gpiozero/gpiozero #4: Name?

bennuttall, 2015-09-15:
I'm getting it ready for PyPI - what do you think of the name? Any suggestions for a better one?

Currently it would be installed with `pip install gpio-components` and the import would be `from gpio_components import LED`.

@waveform80 @lesp @martinohanlon @missphilbin @jrobinson-uk @coding2learn @clivebeale @mattrichardson

jrobinson-uk, 2015-09-15:
Bit long but can't think of a better one

waveform80, 2015-09-15:
The disparity between the dash and underscore is the only objection I'd have (other than the length, but that doesn't bother me so much). Hmm ...

* simple_gpio (after simpleCV)?
* gpiofun (a play on fun as in function)?
* easy_gpio?

Yeah, I'm kinda grasping there... I don't think it *needs* a name change, but it'd be nice to sort out the dash/underscore thing. Pity all the nice short names (GPIO, RPIO, pigpio) are already taken!

bennuttall, 2015-09-15:
Oooooh I've had an idea.

http://pygame-zero.readthedocs.org

GPIO Zero?

lesp, 2015-09-15:
PiZero?

waveform80, 2015-09-15:
GPIO Zero sounds good to me - keeps the GPIO in the name (which makes it obvious what the subject of the package is). I'd recommend doing something similar to the pygame-zero guys though and making the registered package name something without separators, e.g. theirs is pgzero on PyPI. You could do something similarly simple like gpiozero.

bennuttall, 2015-09-15:
Yes, I agree. I read the other day that it's recommended not to have underscores in module names - and read around it and into why. Makes sense - but I hadn't heard of it until then.

I think "zero" shows a match in the philosophy of pygame-zero. I like it.

Let's go with `gpiozero`.

closed 2015-09-15 by bennuttall