gpiozero/gpiozero #5: InputDevice hierarchy doesn't include LightSensor

waveform80, 2015-09-15:
I'm guessing LightSensor doesn't descend from InputDevice because you don't want things like wait_for_input available for it (given the complexities of calculating its value). However, a slightly cleaner way of rejigging it might be:

![InputDevice hierarchy](http://g.gravizo.com/g?
digraph G {
  graph [rankdir=LR];
  node [shape=rect,style=filled,color="%234477bb",fontname=Sans,fontsize=10];
  LightSensor->InputDevice;
  Button->WaitableInputDevice;
  WaitableInputDevice->InputDevice;
  MotionSensor->InputDevice;
  TemperatureSensor;
})

In other words, shove stuff like `wait_for_input` and `add_callback` in a new `WaitableInputDevice` class below `InputDevice` and bung `Button` below that (and everything else, except `TemperatureSensor`) off `InputDevice`. Happy to bung together a PR if you want!

waveform80, 2015-09-15:
Incidentally, the main reason for suggesting this is that it makes tests like `isinstance(some_object, InputDevice)` possible (well, except for TemperatureSensor, but I've got a few ideas on including that in the hierarchy too - just need to test them out, but that'll have to wait until I get my hands on a camjam 2 kit :)

bennuttall, 2015-09-15:
I was thinking about this yesterday. It reminded me of learning about interfaces (it's weird when you finally find a use for that thing you learned about in university). But they're not in Python ([yet](https://www.python.org/dev/peps/pep-0245/)).

What do you gain by making the `LightSensor` and `MotionSensor` classes inherit `InputDevice`? What's in there that's common to the `WaitableInputDevice` besides the `__init__` which varies anyway?

waveform80, 2015-09-15:
It's basically to inherit the `__init__`, and the `is_active` property, and to be able to tell (via the class hierarchy) what's an input, what's an output, or for that matter what's a GPIO component. Incidentally, on the subject of interfaces, Python does have them in the form of abstract base classes (ABCs) - have a look at the [virtual subclass](https://docs.python.org/3/library/abc.html#module-abc) mechanism in the `abc` module for how this works (TL;DR: you can register something as a "virtual" subclass of an ABC even if it's not an actual subclass of it which in turn allows you to test for whether something implements the interface defined by the ABC).

An ABC would certainly be an alternative to this proposal, incidentally - I was just thinking "why bother with the ABC" when you can do it pretty trivially with a bog standard class.

On the subject of what it inherits, admittedly there's not a great deal at the moment because the base class' init is rather simple. However, noting that you've got attributes for `active` and `inactive` I suspected you might be thinking about making the pull-up/down status configurable at some point, and possibly the edge, in which case there'd potentially be more to inherit.

Having trawled through the Explorer HAT code just now, I'm also very taken with the way they're dealing with callbacks (`pressed`, `released`, etc. etc.). Might be worth thinking about nabbing some of those ideas!

waveform80, 2015-09-17:
Oh, and to inherit new read-only pin property added in #9 ;)

waveform80, 2015-09-21:
Just noticed this is effectively done now anyway (#14 does makes LightSensor an InputDevice descendent)

closed 2015-09-21 by waveform80