gpiozero/gpiozero #7: API design

bennuttall, 2015-09-15:
*This comment will be updated as API design is agreed.*

*`xx()` denotes method, `xx` denotes property.*

**LED**:

- `on()`
- `off()`

**Buzzer**:

- `on()`
- `off()`

**Motor**:

- `on()`
- `off()`

**Button**:

- `is_pressed` (is it currently pressed?)
- `when_pressed()` (callback)
- `wait_to_be_pressed()` (wait for edge)

**MotionSensor**:

- `motion_detected`

**LightSensor**:

- `value`

**TemperatureSensor**:

- `value`

waveform80, 2015-09-16:
Might be worth taking a leaf out of the Explorer HAT API for the button and adding `when_released`, `when_changed` (pressed or released) and similar for the wait functions?

waveform80, 2015-09-17:
Oh, and `blink` for the LED class? (again, an idea shamelessly stolen from Explorer HAT :). I can come up with a PR for these if you want to have a play with them before deciding whether they're worth including?

bennuttall, 2015-09-17:
Yes please! As long as features are:

- useful
- common
- uncomplicated
- don't make other use more complicated

then I'm happy.

waveform80, 2015-09-21:
Okay, as of #14 we've got the following, starting with the generic stuff (I've left out derived stuff and just indicated the parent class):

#### GPIODevice

* pin (RO) - The GPIO pin the device is associated with
* is_active - True if the GPIO is active, False if it's not. Descendants automatically compensate for pull-up/down and yes, this applies to outputs as well (it works, so why not?)

#### InputDevice(GPIODevice)

* pull_up (RO) - True if the input's set to pull-up, `False` if it's pull-down. Read-only
* wait_for_input() - basically an alias for GPIO.wait_for_edge. I'm still not entirely sure about having this in the base class: it doesn't play nice with some current descendants and personally I'd rather remove it and replace it with class specific stuff like `wait_for_press` on Button and similar things on other class'.
* add_callback() & remove_callback() - similar to above - I'd rather have class specific stuff like `when_pressed`, etc. I'm also a bit wary of directly using GPIO's callback mechanism. It's not very friendly if you attempt to add multiple callbacks to a particular device (do we want to support this?)

#### OutputDevice(GPIODevice)

* on() - obvious
* off() - obvious

#### Buzzer(OutputDevice)

Doesn't add anything

#### LED(OutputDevice)

* blink() - do we want defaults for on_time and off_time?

#### Motor(OutputDevice)

Incomplete. What do we do about connection to motor drivers? What about bi-directional ones?

#### Button(InputDevice)

Doesn't add anything

#### MotionSensor(InputDevice)

* sample_rate - number of times/second to query the motion sensor (10)
* queue_len (RO) - number of values to average for `value` (5)
* threshold - value above which `motion_detected` returns True (0.5)
* value (RO) - average of the values in the queue
* motion_detected (RO) - True if value > threshold, False otherwise
* partial - if True, you can query `value` and `motion_detected` before the queue has been filled. Otherwise, querying them will block until the queue is charged (False)

Is this too complex? Everything's got defaults so it's still as simple as ever to construct and use but there's a few things I worry about:

* is_active is still inherited here and simply returns the raw state of the sensor. This might be confusing. Should `is_active` be the same as `motion_detected`?
* there's a lot of constructor parameters; instead we could do something similar to the thermal sensor under the covers: have a database (just a dict) of common sensors and relevant settings then permit people to simply specify their model on construction? The thing is you'd probably still need to provide the raw options to allow people to configure sensors we haven't got in the database...

#### LightSensor(InputDevice)

* darkness_time - the length of time the capacitor takes to charge when it's dark (0.01s)
* queue_len (RO) - number of values to average for `value` (5)
* threshold - value above which `light_detected` returns True (0.1)
* value (RO) - average of the values in the queue
* light_detected (RO) - True if value > threshold, False otherwise
* partial - if True, you can query `value` and `light_detected` before the queue has been filled. Otherwise, querying them will block until the queue is charged (False)

More or less the same questions as above.

closed 2015-10-04 by bennuttall