gpiozero/gpiozero #8: pin property accessible?

bennuttall, 2015-09-15:
```python
led = LED(2)
led.on()  # it works
led.pin = 3
led.off()  # error
```

Users can do this. Should I care? @waveform80 

Alternatively, the `pin` property could be prefixed with an underscore.

bennuttall, 2015-09-17:
It could be handy to read this value, but if people think they can write it, they may expect it to work. It won't because the `__init__` does the setup. I don't think it's worth making this work - but is it worth preventing it? If so, how?

waveform80, 2015-09-17:
Easy enough to make pin a read only property - I'll bung together a PR in a
bit (might be an hour or so - just wrestling with mail servers at the mo!)

bennuttall, 2015-09-17:
Great, thanks

commit 1219d99d (2015-09-17): Fix #8

Make the pin property a read-only property
commit 4c2796e0 (2015-09-17): Merge pull request #9 from waveform80/pin-read-only

Fix #8
closed 2015-09-17 by bennuttall