gpiozero/gpiozero #17: Common boards and accessories to add

bennuttall, 2015-09-21:
- [Pi-Traffic](http://lowvoltagelabs.com/products/pi-traffic/)
- [Pi stop](http://4tronix.co.uk/store/index.php?rt=product/product&product_id=390)
- [Fish Dish](https://www.pi-supply.com/product/fish-dish-raspberry-pi-led-buzzer-board/)
- [Pi Liter](http://shop.ciseco.co.uk/pi-liter-8-led-strip-for-the-raspberry-pi/)

Not including things which already have decent libraries, like Pibrella and Explorer HAT.

bennuttall, 2015-09-21:
Anything else?

bennuttall, 2015-09-22:
I've half implemented the MCB - added a `Robot` class for two single-direction motor robots. RTK MCB uses two dual-direction motors.

bennuttall, 2015-10-02:
Properly implemented Motor and Robot, including RTK MCB. Everything on the list is done now. Anything else to add?

bennuttall, 2015-10-12:
Added suggestions from #56

DexterInd, 2015-11-04:
Possible to put in a request for the Arduberry?

DexterInd, 2015-11-12:
[Arduberry](http://www.dexterindustries.com/arduberry/) ?
[GrovePi] (http://www.dexterindustries.com/grovepi) ?

bennuttall, 2015-11-12:
How do users currently use these products in software?

DexterInd, 2015-11-12:
Hey Ben! 

The Arduberry is programmed in the Arduino language; it's a standalone Atmega328 connected to the Pi. Communication through either C or Python on the Pi.  

The GrovePi is mostly Python (though other languages are supported).  The Python library is a single python file [here on github](https://github.com/DexterInd/GrovePi/blob/master/Software/Python/grovepi.py).  

Did I answer your question?

waveform80, 2015-11-14:
I've had a quick skim through the GrovePi's library that @DexterInd linked to; it looks like it's basically all I2C stuff so it's something we might want to add in future (given we've had a request for I2C based ADCs) but I doubt we can get it done for 1.0. Maybe stick it in a separate ticket for post-1.0 boards?

bennuttall, 2015-11-15:
Yeah there are a few components listed here than aren't making the 1.0 release, so that's fine. Something to look at in future.

DexterInd, 2015-11-15:
@bennuttall and @waveform80 , what's the best way for us to participate?  Would it make sense to do a pull request, or would that just muddle stuff up?

lurch, 2015-11-19:
Something for the "Longer-term additions" list...
I dunno how well it'd mesh into the existing API, but it would be fantastic if gpiozero could also be used to provide a uniform interface to I2C IO-expanders, e.g. http://raspi.tv/2013/using-the-mcp23017-port-expander-with-wiringpi2-to-give-you-16-new-gpio-ports-part-3 and also used on boards like https://www.abelectronics.co.uk/products/17/Raspberry-Pi--Raspberry-Pi-2-Model-B/54/IO-Pi-Plus

bennuttall, 2016-02-10:
Closing this - each component should have its own issue

closed 2016-02-10 by bennuttall