gpiozero/gpiozero #23: Introduce toggle method

bennuttall, 2015-09-22:
I'd like to introduce a `toggle` method for any output device which can be turned on or off. If it's on, toggle will turn it off. If it's off, toggle would turn it on. Applying toggle to a group of components would invert their state.

Would it be easier to maintain state within the object or check it using `GPIO.input(self.pin)`?

waveform80, 2015-09-22:
Hmm ... depends how "correct" you want the toggle to be. For example, consider the following naive implementation:

```python
def toggle(self):
    if self.is_active:
        self.off()
    else:
        self.on()
```

Now consider a multi-threaded scenario with an LED (very common in GPIO given the RPi.GPIO library uses threads for callbacks itself, not to mention all the threaded stuff we're doing in gpiozero):

|Thread 1|Thread 2|Comments|
|---------|---------|-----------|
|LED.toggle()|LED.toggle()||
|if self.is_active||Returns False|
||if self.is_active|Returns False|
|self.on||Correct toggle|
||self.on|Incorrect toggle|

So yes, we can implement toggle pretty easily using GPIO.input (this is almost certainly better than using a separately stored state) but we need to be careful about threading effects. In other words we probably want to wrap the whole thing in a `threading.Lock`.

bennuttall, 2015-09-23:
Interesting. Can you determine whether or not it's blinking? I guess if you call toggle while it's blinking, it should stop blinking. (I guess turn it off?)

waveform80, 2015-09-23:
We could determine whether it's blinking by checking whether the blink thread is alive at that moment, but the call to `on()` or `off()` will implicitly kill the blink thread (as it should). So basically it'll read the state of the LED (let's say it's blinking and it's currently lit), then call `off()` (because it's currently on) which will kill the blink thread and switch off the LED (or buzzer or whatever). In other words, the locked implementation should work fine (wait did I push that last night?)

waveform80, 2015-09-23:
Oh I didn't ... right, well I've got a `toggle` implementation that seems to be working nicely. I'll pull in the latest updates from master, get it all tidied up and open another PR for it...

bennuttall, 2015-09-23:
Cool thanks

commit 25b5756b (2015-09-23): Merge pull request #29 from waveform80/output-toggle

Fix #23
commit 7c647ec6 (2015-09-23): Fix #23

Add OutputDevice.toggle method with locking for correct threaded
operation
closed 2015-09-23 by bennuttall