gpiozero/gpiozero #28: button.when_pressed = led.blink causes Exception

bennuttall, 2015-09-23:
I'm trying a few variations of the following code:

```python
button = Button(17)
led = LED(4)

button.when_pressed = led.blink
```

which always causes the following:

```
Exception in thread Thread-10:
Traceback (most recent call last):
  File "/usr/lib/python3.2/threading.py", line 740, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.2/threading.py", line 693, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/local/lib/python3.2/dist-packages/gpiozero/output_devices.py", line 43, in _blink_led
    if self._blink_thread.stopping.wait(on_time):
  File "/usr/lib/python3.2/threading.py", line 422, in wait
    signaled = self._cond.wait(timeout)
  File "/usr/lib/python3.2/threading.py", line 240, in wait
    if timeout > 0:
TypeError: unorderable types: Button() > int()
```

The following similar examples work:

- `button.when_pressed = led.on`
- `button.when_pressed = led.off`
- `button.when_pressed = traffic_lights.on`
- `button.when_pressed = traffic_lights.off`

But any use of `blink` associated with a callback fails. I can't tell what's causing it. Is it just threads falling over each other?

waveform80, 2015-09-23:
Ahh, did you read my long diatribe on IRC the other night about what we want to do about optional parameters in callbacks? I can dig out the log if you missed it all. Basically that's what's happening here. When you assign `when_pressed` it attempts to bind the call to `blink` with a single parameter to see if it's possible, i.e. it tries to bind `blink(self)`. This works fine (at least as far as call binding goes) because there's an optional parameter but when we actually got to run it, the button instance isn't a floating point number of seconds so everything goes "boom".

Anyway, given that it's perfectly reasonable to expect `blink` to work with callbacks I guess that pretty much settles the argument: my initial hunch was wrong and I should rejig the flexible callbacks stuff from #24 so that it attempts to bind the no-arguments version first and settles for that if it works. I'll bung together a PR for this in a bit.

bennuttall, 2015-09-23:
Ahhhhh, of course. That makes perfect sense.

commit 4e330527 (2015-09-23): Fix #28

Re-order the test bindings to prefer using no parameters if possible
closed 2015-09-23 by bennuttall