gpiozero/gpiozero #34: Public beta for teachers

bennuttall, 2015-09-25:
I think we're close to being ready for a public beta, to get some feedback from teachers.

I'm fairly certain all the main stuff is complete, and perfectly usable. Some of the sensor stuff needs more work, and only basic motor functionality has been added so far, but examples using LEDs, buzzers and buttons will be fine.

I need to double check there are no missing docstrings (particularly in inherited methods), and I need to document the usage of all the completed components, but otherwise I think we're set.

I'll also fill in more info on the google doc - that'll be handy for teachers. And create some recipes and example projects - and look at converting more of the code in our resources as further examples.

@waveform80 any lingering thoughts?

waveform80, 2015-09-26:
I'm going to have a crack at #27 this weekend.

On motors: I'll grab an RTK MCB and see what I can get done next week, but it probably won't be much given time constraints.

On docstrings: I don't *think* there are any missing docstrings but then I'm not entirely convinced we actually need overridden docstrings to the extent that we've got them. I know the idea is to keep the library as easy to use as possible but I'm not convinced people really need "Turns the LED on" instead of "Turns the device on" in the subclass, for example (or, if that really is a problem - imagine the issue we'll have if we ever change a docstring in a base class and don't propagate that change to *every* subclass leading to inconsistent / incorrect docstrings).

If I was going to change one thing at the moment, it'd be to strip out all that mostly duplicated stuff and just see how it pans out initially (my hunch, as noted on #10, is that most people never go near the API docs and those that do tend to be more than capable of figuring out "device" means "LED" or whatever the context is). But that's just my "long term maintainability" imp yelling in my ear - your call if you want to change that or not.

Otherwise, I'd say it's definitely ready for a bit of real-world testing!

bennuttall, 2015-09-26:
I think you're right, let's drop the duplicated docstrings.

closed 2015-09-29 by bennuttall