gpiozero/gpiozero #36: Callbacks not working for button when pulled down

bennuttall, 2015-09-28:
This example works:

```python
button = Button(14)

button.when_pressed = hello
```

But this example seems not to work:

```python
button = Button(pin=14, pull_up=False)

button.when_pressed = hello
```

waveform80, 2015-09-28:
It does work but you've got to wire the circuit differently. When the GPIO is pulled high it's a *source* of current so you wire the circuit like this:

```

+--------+
|        | GPIO14
|        |
|        +---------+
|   Pi   |         |
|        |         |
|        |         |
|        |         o
+----+---+          \
     |               \
GND  |             o
     |             |
     |             |
     |             |
     +-------------+
```

Nice and simple: you push the button and current flows from the pulled-high GPIO and sinks in the ground pin. Because current is flowing, the GPIO pin is "high". When you release the button, the GPIO pin goes "low" because no current is flowing, but it's still pulled-up so it has the *potential* to flow when you push the button.

Now consider if it's not pulled up. You push the button but no current flows because the GPIO is pulled down. There's no potential there. So you need to re-wire your circuit like this:

```
     +-------------+
     |             |
     |             |
3v3  |             o
     |              \
+----+---+           \
|        | GPIO14  o
|        |         |
|        +---------+
|   Pi   |
|        |
|        |
|        |
+--------+

GND
```

Now we've got potential from the 3v3 pin and when you push the button, current flows and sinks in the pulled-down GPIO pin, raising it "high". Let go of the button, no current flows, GPIO pin goes "low".

So, it does work - just requires a different circuit. Which one is the more obvious? Probably the pulled-up variant which is why it should probably stay as the default on `Button`. However, pulled down is definitely "safer" in as much as it guarantees no current flows from the GPIO so it should probably remain the default on the `InputDevice` base class.

bennuttall, 2015-09-28:
Oh, of course. My mistake!

closed 2015-09-28 by bennuttall