gpiozero/gpiozero #37: Add keep alive functionality

bennuttall, 2015-09-28:
I've been doing all my testing in an IPython shell and it's great fun. However, I just tried to run an example in a file and nothing happened because there was nothing keeping the program alive.

waveform80, 2015-09-28:
The following works nicely:

```python
from gpiozero import Button, LED
from signal import pause

btn = Button(26)
led = LED(16)

btn.when_pressed = led.on
btn.when_released = led.off

pause()
```

So ... we could either write script examples like this or pull `pause` into gpiozero and alias it as something descriptive ... `run_forever`, `run_until_killed`, `wait_for_ctrl_c` (not so good under Idle), erm ... `keep_alive`?

Personally I think I'd just write examples like the above and not bother with the aliasing. `pause` is fairly descriptive in and of itself, and this makes people aware of one of the standard library's many modules (signal) which they may find useful under other circumstances (for example, the Explorer HAT library has a similar callback mechanism but doesn't explicitly provide a function for pausing the main thread so this method would be equally applicable there).

However, one word of caution is that `pause` terminates after *any* signal, not just `SIGTERM` or `SIGINT`. For example, `SIGSTOP` (Ctrl+Z) would terminate the process too. That's probably not too much of a concern (I doubt kids / newbies know about backgrounding processes in bash :) but I wonder if other signals get used in environments like Idle. I'll have a quick play and see how it works.

bennuttall, 2015-09-28:
`from signal import pause` is nice and neat. I'm torn between throwing that into the docs and adding `keep_alive = pause` into the lib. I'm leaning towards the former.

waveform80, 2015-09-29:
I've had a play with this under Idle now and it seems to work nicely - pause works happily and just waits around until killed with the Shell Restart (Ctrl+F6) thing which is exactly what we want.

bennuttall, 2015-09-29:
Ok I'll add that to the docs where appropriate

closed 2015-09-29 by bennuttall

bennuttall, 2015-09-29:
Another thought just occurred to me: an alternative method for keeping a program alive would be to use the built-in function `input()` with no argument (or `input("Press Enter to end")`).

It would save having to import another library.

However, `input` varies between Python 2 and 3. Would work perfectly in Python 3, but Python 2's `input` expects an integer to be entered and throws an error if blank. The equivalent in Python 2 is `raw_input` but that adds a complication being different. Shame, really.