gpiozero/gpiozero #38: Can't recreate a Button object

bennuttall, 2015-09-28:
Running the following code in a shell:

```python
>>> button = Button(15)
>>> button = Button(15)
```

The second line causes a `RuntimeError` raised by `GPIO.add_event_detect()`

Oddly, trying a couple more times, it eventually lets you.

I also managed to get a Segfault doing this once too many times! (Probably a `RPi.GPIO` problem, not `gpizero`)

waveform80, 2015-09-28:
Yup, welcome to the joys of unpredictable garbage collection. Here's what's going on:

* You construct the first Button and that calls RPi.GPIO to setup GPIO 15 as an input. The result is bound to the name `button`.
* You construct another Button. At this point, the name `button` is still bound to the first one so the first Button instance hasn't been destroyed yet, so its `__del__` method hasn't run, cleaning up the callback.
* I'm assuming that at some point in your playing you get rid of the first Button instance by assigning something else to the name `button` and leaving no references to it. At that point the garbage collector is free to kill it off (running its `__del__` method in the meantime). Of course, there's no guarantee it actually *does* kill it at that point - it just *can*.
* Then later on you get to construct it again.

I'd suggest we ought to add a method for explicitly closing down the GPIO instance (like the common `close` method found in file objects, sockets, the PiCamera class, and just about anything where it can be useful to manually control its life-span, or at least its usable life-span), then simply re-write `__del__` to call that method (and squash any exception raised if it's already been called).

bennuttall, 2015-09-28:
Can we not just remove the event detection before adding a new one?

waveform80, 2015-09-28:
Firstly, how do you know there's one to remove (although you could just do EAFP here)? Secondly, the method that removes event detection removes *all* existing handlers for that pin (you can't selectively remove callbacks). That's *probably* not an issue right now (because we're not exposing the `add_event_callback` mechanism directly) but this would interfere with any attempt to expose it in future (not suggesting we should, but if we ever want to go that route it'll be an issue).

To make the case a bit further: consider the case where someone explicitly wants to clean-up a particular GPIO port (for whatever reason). In RPi.GPIO they can do it (GPIO.cleanup(14)) but at the moment in gpiozero the only method is to try and  get rid of all references to the object holding that port and then hope the garbage collector collects the object. That's not a great interface :)

PR #39 (2015-09-29): Explicit cleanup
This is a rather complex PR which partially (but not completely) attempts to fix #38. There are several issues here which include:

* Tracking object lifespan in a garbage collected, dynamic language is *hard*. Hard enough that I'd never expect a newbie programmer to understand it (in the case of GPIO stuff it involves figuring out when objects are being kept alive because a callback they implement is referenced by a chunk of external C code).
* Even if you can track the lifespan you can't force the garbage collector to do what you want. You may think the object can be killed  [... cut]
closed 2015-09-29 by bennuttall