gpiozero/gpiozero #40: Should RGB LED values be 0-100 like PWM or 0-255?

bennuttall, 2015-09-29:
The `RGBLED` class I implemented yesterday comprises of three values, for red, green and blue. The way PWM works is that you set the duty cycle from 0 to 100, from off to dim to bright. If the interface allowed the user to pass in values on a scale from 0 to 255, this would be more compatible with usual colour codes like in CSS and in the Sense HAT library.

bennuttall, 2015-10-02:
Ok I think I'll leave the PWM class an a 0-100 scale and make the RGBLED class use 0-255 and convert accordingly.

waveform80, 2015-10-03:
Did you get the comments I made on IRC the other day about a 0-1 scale? Wasn't sure as your connection dropped halfway through. If you didn't like the idea, no worries.

bennuttall, 2015-10-03:
I'm not sure. They both make sense but my gut says 0 - 100 is a simpler interface.

waveform80, 2015-10-03:
Hmm, my main argument in favour of 0-1 is that it doesn't falsely imply a precision. 0-100 implies that 101 discrete steps are supported by PWM (I've no idea if that is or isn't the case, but the API implies it). Similarly for 0-255, 256 discrete steps are implied. Meanwhile 0-1 gives the impression of "some analog value between nothing and 1" (which again probably isn't true, but it's also non-specific).

On an unrelated note: I wouldn't worry about compatibility with the Sense HAT library because under the covers that doesn't support 0-255 anyway (it's an RGB565 framebuffer so it's 0-31 steps for red and blue, and 0-63 for green - and even then I think there's still some bit fiddling going on in the hardware below that).

bennuttall, 2015-10-04:
Actually now I think about it, 0 being off and 1 being on, 0-1 makes sense. Although that's not how we provide the interface. Let's go with it.

However, I think in the case of RGB LED, it's much more common to provide values up to 255.

commit 493b4e1c (2015-10-05): Change RGBLED interface from 0-100 to 0-255, close #40
closed 2015-10-05 by bennuttall