gpiozero/gpiozero #41: MCP3008 / Potentiometer

bennuttall, 2015-09-30:
I've incorporated @martinohanlon's MCP3008 class, and added a couple of examples to the recipes in the docs, including a cool one for controlling an RGB LED with 3 pots.

`MCP3008` could be used for a range of analogue inputs. Is this sufficient an interface, or should we provide wrappers for specific components?

martinohanlon, 2015-10-03:
How about creating a super class of ADC which describes the interface for an ADC of which an MCP3008 is one of them. You could then create wrappers for analog components (e.g. potentiometer) and you pass a reference to your ADC when you instantiate it 

myadc = MCP3008(channel = 2)
mypot = Potentiometer(myadc)

or

myadc = ADS1115(address=0x48)
mypot = Potentiometer(myadc)

waveform80, 2015-10-03:
Did you mean the `open` method to be called from `__enter__`? That implies that either users *have* to use the class as a context manager, or that they have to call `open` manually after construction. The more usual pattern is to perform initialization implicitly on construction, have `__enter__` return self, and then call `close` from `__exit__`.

waveform80, 2015-10-03:
BTW - I haven't had a chance to play with ADC or pots generally yet (at least in connection with gpiozero), but @martinohanlon's suggestion looks very nice. My one concern at the moment is that people may wish to use pots in a similar manner to the LDR class (in series with a capacitor, time the charging and average it for a value) but we could make this compatible with the suggestion: derive from SmoothedInputDevice, detect if pin is a plain integer and treat it in a similar manner to the LDR circuit if so; alternatively if it's a derivative of some generic ADC class, act accordingly - e.g. reduce queue_len to 1 (in which case I'd rename the `read` method to `_read` and let the rest of the SmoothedInputDevice machinery take over).

martinohanlon, 2015-10-03:
When I wrote the MCP3008 class, I wanted to give the consumer the option of using with if they wanted too, so you could consume it as either

    with MCP3008() as myadc:
        myadc.read()

which would have the benefit of opening and closing the SPI channel 'automatically'

or you could implement it as:

    myadc = MCP3008()
    myadc.open()
    try:
        myadc.read()
    finally:
        myadc.close()

I vaguely remember that for one of the use cases I had the above was more 'useful' for me, but I cant think why!

This was my first 'attempt' at creating a class which used the context manager, if there is a better way lets change it.

waveform80, 2015-10-03:
It's definitely useful to have both methods; it's just slightly odd to have `open()` as something explicitly called if it's not something that's typically called repeatedly during the object's lifetime (I'm making an assumption here that it's not but do correct me if I'm wrong!). For example, with files:

```python
with io.open('some_file.txt', 'rb') as f:
    s = f.read()
```

is equivalent to:

```python
f = io.open('some_file.txt', 'rb')
s = f.read()
f.close()
```

So in this case, the acquisition of the resource (in this case the file descriptor) is performed by the constructor, `close()` handles releasing the resource, and the context manager ensures that `close()` is called implicitly. This is the case where the resource is something used once in the object's lifetime. PiCamera is equivalent to this case:

```python
with picamera.PiCamera() as camera:
    camera.capture('foo.jpg')
```

is equivalent to:

```python
camera = picamera.PiCamera()
camera.capture('foo.jpg')
camera.close()
```

Contrast this with the case of something which is acquired multiple times during the owning object's lifetime like a threading lock:

```python
lock = threading.Lock()

with lock:
    # do stuff protected by the lock
```

is equivalent to:

```python
lock = threading.Lock()
lock.acquire()
try:
    # do stuff protected by the lock
finally:
    lock.release()
```

So in the above case, a separate acquisition method is publicly exposed because it's likely to be required repeatedly during the lock object's lifetime. My assumption is that the MCP class, representing something that's physically wired to the Pi, is likely to be a "one-shot" class in as much as it's constructed, acquiring the resource (bus, device, channel) associated with it, performs operations with that resource (read/etc.), and then dies with the script.

If that assumption is correct then for the sake of consistency with convention we ought to move the open() method's operations into the constructor and do away with the method. Otherwise, it should be left as is.

bennuttall, 2015-10-04:
Ok we have two questions here:

1. Should we provide a generic ADC interface?

    My gut feeling is yes, but I don't know what other ADC components there are, and how they work. If they're all 0-1023 values then I don't see any reason to - and what we have should just be called `ADC` or `AnalogueInput`.

1. How should the MCP3008/potentiometer interface be provided?

    I like Martin's context manager but my only concern is that it's unlike all the other interfaces. Would it be possible to provide an interface similar to our other inputs?

bennuttall, 2015-10-07:
Any thoughts on this? @waveform80 @martinohanlon?

waveform80, 2015-10-13:
The range of values returned by an ADC is determined by it's "[effective number of bits](https://en.wikipedia.org/wiki/Effective_number_of_bits)" so the MCP3008 returns 0-1023 because it's a 10-bit unit (2^10 being 1024). Other devices will have different ranges.

I probably ought to get hold of a few of these and experiment with them to see whether I can come up with a decent generic interface. It seems like a lot of them support SPI, so that's a good start but we're definitely going to need a configurable range and I'm very tempted to have its output normalized to the 0 to 1 float range to keep things consistent with everything else in the library.

bennuttall, 2015-10-13:
I'm currently trying to get two working at once but no luck just yet.

I wired up according to [this](http://www.raspberrypi-spy.co.uk/2013/10/analogue-sensors-on-the-raspberry-pi-using-an-mcp3008/mcp3008_tmp36/#prettyPhoto/0/)

I'm in favour of a normalised 0-1 return value.

What do you think about the interface being very different from the others? How about an `AnalogueInputDevice` class with a `value` property which runs the `MCP3008`'s `read()` method?

waveform80, 2015-10-13:
Basically, I'd suggest `AnalogInputDevice` (American spelling) as a base class which assumes an SPI input and thus takes `bus`, `device`, and `channel` parameters to its constructor (like `MCP3008`), along with `bits` (number of bits the device outputs which in turn will determine its range) and possibly (i.e. if required, to be determined by experimentation or user demand) values to bias the conversion. There should be a "private" `_spi` property containing the `SpiDev()` object used for communication with the device, a "private" `_read` method which returns the raw value but which in the base class just raises `NotImplementedError` (effectively making it an abstract base class) because what it does will depend on the specifics of the ADC's communications protocol. Finally, it should also implement a public `value` property (same as MotionSensor, LightSensor, PWMOutputDevice, etc.) which provides the raw value normalized to the 0-1 range (we can do this in the base class because we know the number of `bits` provided by the ADC).

`MCP3008` (and other ADCs) could then be a sub-classed off this with the `bits` parameter removed from the constructor (because we know it's 10 in the case of MCP3008) and `_read` overridden to handle the specifics of the SPI communication.

Personally I'd also merge `open()` into the base constructor (so it's implicit - I don't get the impression these are devices that are opened and closed repeatedly in their lifetime), and keep the context manager implementation basically as is (except remove `open` from `__enter__` as it wouldn't be needed there). I'd probably also tweak bus, device, and channel to be read-only properties (is there a use-case for permitting these to be modified at run-time?).

Oh, when you say you're trying to get two working at once - you mean two channels of the same MCP3008? Hmm ... I wonder if you're allowed two `SpiDev` objects which both have the same bus and device to exist at the same time? If not, I may have to get a bit cunning in the base constructor to let instances share `SpiDev` instances where required (don't want to expose the user to that complexity when we don't have to).

bennuttall, 2015-10-13:
Ok cool. Sounds good!

> Oh, when you say you're trying to get two working at once - you mean two channels of the same MCP3008?

No, that works fine. I mean two MCP3008s providing a total of 16 channels. @martinohanlon says it works if you use a different clock pin so I'll try that.

bennuttall, 2015-10-13:
Ooh got it working! 16 analogue inputs!

bennuttall, 2015-10-13:
Actually no...

waveform80, 2015-10-13:
Can you try two inputs on a single MCP3008 with two different objects simultaneously? Just want to check that two `SpiDev` objects can indeed address the same device at once.

bennuttall, 2015-10-13:
Yes that works.

bennuttall, 2015-10-13:
Ah - got it working now. Was just the fact I'd previously configured the pins as outputs.

waveform80, 2015-10-15:
Right, been doing a bit of work on an `AnalogInputDevice` base class. A few changes I've made: I removed `bus` from the constructor as there's only one (user accessible) SPI bus on a Pi so there's little sense bothering the user with it. There's two accessible devices though so that parameter needs to stay in. I've altered the constructor to immediately open the device, added an abstract `_read` method, a `value` parameter which normalizes the `_read` result, added a `bits` parameter to the abstract base class and removed it in the concrete subclasses.

There seems to be no guard against two instances opening the same SPI device and I'm a little worried about threaded interactions there, but I figure we'll leave that until (/if) we actually encounter problems before I dive into mutexes and locking shared instances.

I'll PR this in a mo and then people can have a play with it (for the sake of competeness I've added in both MCP3008 and MCP3004 base classes given the latter is a trivial specialization of the former). I haven't got any ADCs to play around with here at the moment, so this is currently untested code - do let me know if there's any problems!

bennuttall, 2015-10-15:
Excellent - thanks Dave! I'll give it a try.

closed 2015-10-15 by waveform80

lurch, 2015-11-04:
Sorry to comment on a closed issue, but has any thought been given to supporting i2c-connected ADCs?
e.g. http://quick2wire.com/products/quick2wire-i2c-analogue-board-kit/
or http://www.hobbytronics.co.uk/adc-i2c-slave
or https://www.adafruit.com/products/1085
or https://www.abelectronics.co.uk/p/17/ADC-Pi-V2---Raspberry-Pi-Analogue-to-Digital-converter

I own one of the quick2wire and abelectronics boards, if you get to a position where you need someone to test.

waveform80, 2015-11-04:
Can't say any thought's been given to this yet. I don't think either of us is an electronics expert (I'm barely even amateur and hence wasn't even aware of the existence of such things!) - but if we can support it, then why not.

@bennuttall this will add a dependency on the SMBus module, but this is already packaged for python 2 and 3 on Raspbian so I doubt that's a big deal. We may also want to consider making a couple of base classes below AnalogInputDevice, one for SPI devices, one for I2C depending on how much similarity there is between the methods of reading these components (I've only briefly skimmed the links provided above so I'm not sure yet). Probably worth a new ticket though?

pcopa, 2015-12-18:
The MCP3008 is obsolete. I got the MCP3208 cheaper and it's 12 bit. The code is practically the same, only add 2 bits.
So I would like to propose adding a MCP3208 option.

bennuttall, 2015-12-22:
MCP3208 looks great! I'll see if I can get hold of a few and get it added. Thanks

pcopa, 2015-12-28:
Just found out today that there is also a 13 bit version: MCP3308.  And they all have a 4 Channel version MCPxx04 too.

pcopa, 2015-12-28:
Just another thought;  The new LED chips and strips have added a 4th channel "White", so its RGBW now.
Adding an additional line to the recipe would make the code more universal.

lurch, 2015-12-28:
> Just found out today that there is also a 13 bit version: MCP3308. And they all have a 4 Channel version MCPxx04 too.

I just did a google, and found https://ethandicks.wordpress.com/2013/06/19/adding-analog-input-to-raspberry-pi/ which says there's an MCP3002 too.
And http://www.microchip.com/ParamChartSearch/chart.aspx?branchID=11021&mid=10&lang=en&pageId=79 lists even more devices.

                 | 10-bit       | 12-bit         | 13-bit
--------------|---------------|---------------|--------------
1 channel | MCP3001 | MCP3201 | MCP3301
2 channel | MCP3002 | MCP3202 | x
4 channel | MCP3004 | MCP3204 | MCP3302
8 channel | MCP3008 | MCP3208 | MCP3304

The 13-bit MCP33xx devices are all usable as differential (measure the **difference** between two arbitrary voltages) or single-ended (measure the level of a voltage above ground) inputs, so *may* not be quite as drop-in compatible as the MCP30xx and MCP32xx devices? \*shrug\*

pcopa, 2015-12-28:
@lurch thanks for the research.
Just a small editorial change. Shouldn't the last two entries in the last column be 04 and 08?

lurch, 2015-12-28:
> Shouldn't the last two entries in the last column be 04 and 08?

Nope, check the [datasheet](http://ww1.microchip.com/downloads/en/DeviceDoc/21697F.pdf) ;-)
Because they're designed to be differential ADCs, the MCP33xx part-number refers to the number of differential channels. Each differential channel can alternatively be used as two single-ended channels. (although I don't understand ADCs well enough to know why the MCP3301 can't be used as two single-ended inputs)

IMHO it's probably easiest to just ignore the MCP33xx devices, and focus on the MCP30xx and MCP32xx.

lurch, 2015-12-28:
And just for completeness / comparison, I've just looked up the prices of the DIP (i.e. breadboard-able) versions of the chips on https://uk.farnell.com
For the MCP32xx devices I've gone for the more-accurate 'B' version rather than the 'lower spec' C version (although I dunno how much difference this makes in the real-world). Farnell currently don't stock the MCP3004 in a DIP package. All parts are available in single quantities.

Part           | Pre-VAT price (as of 2015-12-28)
--------------|---------
MCP3001 | £1.13
MCP3002 | £1.22
MCP3004 | not stocked
MCP3008 | £1.52
MCP3201 | £1.80
MCP3202 | £1.98
MCP3204 | £2.71
MCP3208 | £3.58

waveform80, 2016-01-31:
@lurch any chance you could open another ticket to remind me about those extra chips? I know I'll forget otherwise, and my head's full of pin-rebase stuff at the mo!

lurch, 2016-02-01:
> @lurch any chance you could open another ticket to remind me about those extra chips?

Done #162

commit a347a2ca (2015-10-15): Fix #41

Add `AnalogInputDevice` base class and extend `MCP3008` (and `MCP3004`)
classes from it.
commit e1868cdc (2015-10-15): Merge pull request #73 from waveform80/analog-input-device

Fix #41