gpiozero/gpiozero #44: Deb packaging

waveform80, 2015-10-03:
We should get on with Raspbian packaging fairly quickly. I want this on the image for the next Birmingham picademy, and I'd like to include instructions on the worksheet for teachers to install this on any other Pi's they have with minimal hassle. I can handle this if you wish.

bennuttall, 2015-10-03:
Yes please - go ahead

waveform80, 2015-10-05:
Argh ... spidev and w1thermsensor aren't packaged for Raspbian. I'll have to make those local imports and come up with some "friendly" error telling the user how to install them in the event they're not present

waveform80, 2015-10-05:
(or package them myself, but I really don't have the time to devote to that)

bennuttall, 2015-10-05:
Hmm. Maybe we should look at packaging them. I'll ask when I'm back in the office.

timofurrer, 2015-10-07:
@bennuttall Is this the issue why you need the w1thermsensor module as package in the repos?
I'll definitely gonna work on it ...

bennuttall, 2015-10-07:
Yes - thanks!

waveform80, 2015-10-09:
Quick question before I redo the PR for deb packaging - I'm assuming the w1thermsensor package is going to be packaged as python-w1thermsensor and python3-w1thermsensor, and that spidev will be packaged as python-spidev and python3-spidev in accordance with the usual Debian convention? Just need to check for the deps in the control file

bennuttall, 2015-10-09:
I would think so. @XECDesign?

XECDesign, 2015-10-09:
Yeah, that's right.

timofurrer, 2015-10-09:
Yes - for w1thermsensor the same ..

waveform80, 2015-10-09:
Good stuff - I'll bung in the PR in a bit

bennuttall, 2015-10-10:
Reopening until the package is released!

waveform80, 2015-10-10:
Heh - good point :)

timofurrer, 2015-10-12:
Use the following commands to create the python 2 and python 3 packages:

For python 2:
```bash 
python setup.py --command-packages=stdeb.command bdist_deb
```

For python 3:
```bash
python3 setup.py --command-packages=stdeb.command bdist_deb
```

Do you need something else in order to include it to the repositories?

waveform80, 2015-10-12:
The machinery for making the deb package should all be in there after PR #54, we're just fixing a few other things before the next release at the moment. The machinery is based on stdeb's output but with a few tweaks that I've built up over the years to make certain things easier (most of it's in the Makefile and handles things like launching dch to update the changelog properly and in the fuller versions that I use on picamera and picraft, building the separate docs package automatically, integrating it with docs-base, and auto-uploading everything to PyPI and the Raspbian maintainer via dput)

bennuttall, 2015-10-25:
I didn't use this for the 0.9 release as I needed to get it out - but will try by the next release.

waveform80, 2015-10-26:
No prob - let me know if you have any issues. I can always run it if you like as I'm set up for signing packages and Serge's got my public keys on the Raspbian dput servers (which is why I can automate the whole thing from package building to uploading to PyPI and Raspbian)

lurch, 2015-11-04:
Will the python-w1thermsensor, python3-w1thermsensor, python-spidev and python3-spidev packages also be added to the Wheezy repositories? Or will they remain Jessie-only?

waveform80, 2015-11-04:
That would be the "proper" solution here (and I suspect the easiest though
I'm not au fait with the particulars); I haven't heard the verdict on that
yet though.

bennuttall, 2015-11-04:
That would make things a lot easier. We do want people to upgrade to Jessie, but I don't like the idea of this just not working in Wheezy. It's possible to supply separate instructions for Wheezy but that gets complicated. 

@XECDesign do you think we could make Wheezy packages for this and its dependencies?

XECDesign, 2015-11-04:
If someone can confirm that they 'just work' as they are, I don't see why not.

bennuttall, 2015-11-04:
They should! Send me some debs and I'll test them. Thanks

XECDesign, 2015-11-04:
http://archive.raspberrypi.org/debian/pool/main/w/w1thermsensor/python-w1thermsensor_0.2.2-1_all.deb
http://archive.raspberrypi.org/debian/pool/main/w/w1thermsensor/python3-w1thermsensor_0.2.2-1_all.deb
http://archive.raspberrypi.org/debian/pool/main/s/spidev/python-spidev_2.0~git20150907_armhf.deb
http://archive.raspberrypi.org/debian/pool/main/s/spidev/python3-spidev_2.0~git20150907_armhf.deb

waveform80, 2015-11-16:
Right, packaging is tested and works nicely (on jessie - it'll work on wheezy as well when/if the spidev packages are back-ported there) so I'll close this for release

commit ac5eb7ec (2015-10-10): Merge pull request #54 from waveform80/raspbian-packaging

Fix #44
commit c8105c8d (2015-10-10): Fix #44

Implement native Debian packaging (control files are part of the source
repo) with Makefile to simplify the release procedure.
closed 2015-11-16 by waveform80