gpiozero/gpiozero #47: Motor interface has unconsistent optional arg

bennuttall, 2015-10-05:
I gave the `Motor` class's methods an optional argument for the number of seconds to be on for, e.g:

```python
motor.on(5)
```

is equivalent to:

```python
motor.on()
sleep(5)
motor.off()
```

Should we remove this, or add a similar interface to other classes? e.g. `LED`

waveform80, 2015-10-05:
Hmm, my only concern would be that it's different to the frequently used explorer HAT interface which uses the argument to on (or the forward alias) for the speed of the motor. Do we have a means of controlling the speed of the motor (as far as I'm aware, several motor controllers do provide this capability).

bennuttall, 2015-10-05:
How does speed work with Explorer HAT? PWM?

waveform80, 2015-10-06:
[Yup](https://github.com/pimoroni/explorer-hat/blob/master/library/explorerhat.py#L175)

bennuttall, 2015-10-06:
Ok I'll have a go when I get chance, see if it works.

bennuttall, 2015-10-07:
I made Motors use PWM which can be given a speed parameter rather than number of seconds to be on.

closed 2015-10-07 by bennuttall