gpiozero/gpiozero #49: RPi.GPIO Problems on Raspberry PI 2

radames, 2015-10-06:
I've been facing many problems using the RPi.GPIO library at the PI 2, so far the solutions was to use the [gpio lib](https://pypi.python.org/pypi/gpio/0.1.2) that uses the standard linux sysfs interface to deal with it. Wouldn't be a good approach to use it on the gpiozero lib?

waveform80, 2015-10-06:
I can't say I was aware of the gpio lib (and I thought I knew all of them: RPi.GPIO, RPIO, and pigpio) but it's an interesting approach. After running into a couple of issues myself I'm sorely tempted to switch to something more reliable but there's several issues in gpio lib's case:

1. It's not packaged for Raspbian (given it's relatively minimal and based on simple FS manipulations I'd be more tempted to just duplicate the bits required in gpiozero rather than package it manually, but still...).

2. No callback functionality (this is rather crucial for large chunks of gpiozero). I'm not aware of a good way of doing this via sysfs (by good way I mean a way that minimizes the risk of "missing" events even with the main thread busy).

3. No PWM functionality (as far as I'm aware this can't be done via sysfs).

4. No pullup capability; this is required for supporting certain addon boards and buttons. The gpio lib code notes sysfs doesn't support this.

Ultimately I'd love to use RPIO as an alternate backend (simply because I think its PWM support is superior to RPi.GPIO's) but its Pi 2 compatibility PR has been languishing for ages (not a criticism of those working on it; I wish I had more time to pitch in and lend a hand myself!).

bennuttall, 2015-10-06:
What problems? It's installed by default in Raspbian and although it's an odd and unpythonic API, the common examples do work and it's pretty easy to get going. Can you give specifics?

radames, 2015-10-07:
@bennuttall  I've been some issues to make RPi.GPIO python library working on my PI's 2, not sure yet why, I've try to use the version from `pip install rpi.gpio` and from the `apt-get install python-rpi.gpio`, both are not working, basically as root and using the BCM mode trying to turn on and off a pin. I've ending up trying another library and it's working the `gpio` and works fine. This command line tool also works pretty well [wiringpin](http://wiringpi.com/the-gpio-utility). Seems like others had the same [problem](https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=99281&start=75)

@waveform80 has good points, my suggestion was simply because I've saw the post about the RPI.GPIO problems on the PI2 and I was wondering is there are other solutions. I'm still trying to debug the problem maybe this is the [cause](http://sourceforge.net/p/raspberry-gpio-python/tickets/102/)

Anyway the gpiozero seems amazing and would be great to have a stable GPIO behind it, specially for the beginners who will struggle to understand what's happening behind.

bennuttall, 2015-10-07:
Are you using Raspbian? It's installed by default

closed 2015-10-07 by bennuttall

radames, 2015-10-07:
Yes I'm using Raspbian Wheezy