gpiozero/gpiozero #50: Heisenbug in add_event_detect

waveform80, 2015-10-07:
The one issue I've come across in practice with RPi.GPIO (which annoyingly happened during the demo too the other day!) is that sometimes (not often - it's a Heisenbug) the library fails to add edge detection when requested. It happens particularly with Button and LightSensor which both use `add_event_detect`. I haven't yet tied down if there's any pre-requisites to re-produce but it's something I need to look into as when it manifests the only option is to shut down the interpreter and try again.

So, first step: mitigation. Ensure all resources allocated in all constructors are cleaned up in the event the constructor fails (in particular cleaning up the allocation of the GPIO pin and pull-up). Second step: dig into the C source code and find out what on earth is causing `add_event_detect` to actually fail on rare occasions. I'm leaving this as the second step as it's going to be Hard (tm); probably some subtle race condition or something like that.

waveform80, 2015-10-10:
Mitigation is now done - just digging into the C to see if I can figure out the root cause...

bcroston, 2015-10-11:
Possibly linked to one of these:
https://sourceforge.net/p/raspberry-gpio-python/tickets/112/
https://sourceforge.net/p/raspberry-gpio-python/tickets/88/
https://sourceforge.net/p/raspberry-gpio-python/tickets/85/
https://sourceforge.net/p/raspberry-gpio-python/tickets/81/

waveform80, 2015-10-11:
@bcroston thanks for the links. I don't *think* it's related to 81, 85, or 88 as it doesn't cause a segfault and it doesn't prevent future epoll events. It might be related to 112 but it's too vague a report for me to be sure at the moment.

I have traced the issue to line 725 of py_gpio.c so far ("Failed to add edge detection") but I haven't gotten further yet; looks like there's a few places in the called function (`add_edge_detect`) that can return a result that leads to that error so I'll need to trace a bit further - out of time tonight though (unfortunately I still haven't come up with a decent reproduction method so it takes a while to test!)

waveform80, 2015-10-12:
Damn ... I can't seem to reproduce this any more. Oh well, it's worth having the mitigation stuff in anyway so I'll push that and close this for the time being. Maybe a recent kernel update fixed or changed something.

waveform80, 2015-11-14:
Re-opening after reproduction by myself and others in #99

waveform80, 2015-11-14:
This isn't going to get done for 1.0 (partially because it's an upstream issue, but I'll see what I can do post 1.0

bennuttall, 2015-11-15:
The following code (nasty as it is), does seem to work:

```python
from gpiozero import Button

while True:
    try:
        btn = Button(14)
    except RuntimeError:
        pass
```

but can we do something similar in init?

bennuttall, 2015-11-15:
Agh that's wrong ... 1 sec

bennuttall, 2015-11-15:
fixed

waveform80, 2015-11-15:
Possibly (it's the adding of the event detection callback that does it, so we could retry that indefinitely), but ideally I want to get this fixed in the underlying library - it just shouldn't happen at all.

waveform80, 2015-11-16:
Notes for my reference:

* After patching RPi.GPIO to return unique error conditions for several scenarios I've tracked down two causes of this issue (so there's a couple of things to fix here - but I'm not sure about the root causes yet so I'm not sure how hard either is yet)
* One cause is a leak in RPi.GPIO's thread handling. The event thread is stopped and started as necessary, but the thread only calls `pthread_exit`. To clean up the thread's resources it's necessary to call `pthread_join` from another thread *after* the thread has been cancelled or exited. This mean that if one repeatedly starts and stops event detection one can pretty quickly exhaust the available resources and the pthread_create call returns EAGAIN (which isn't retried and is treated as a permanent error - although that's probably reasonable in the circumstances).
* Another cause is a race condition somewhere in the chain of calls under `new_gpio`. I haven't tracked down the root cause here but my gut instinct is that it's either something to do with the arbitrary delay in `gpio_export` or the lack of locking on gpio_list. `new_gpio` mutates it and `poll_thread` in a background thread enumerates it; neither uses mutexes so it's possible there's a nasty race condition here. Although if that's the case I'd expect a segfault rather than the error we're seeing (that said I've seen an unexplained segfault in gpiozero on one occasion so maybe this occurs but extremely rarely).

bcroston, 2015-11-16:
Thanks for tracking that down - I'll take a look and see if I can make a fix at some point this week.

waveform80, 2015-11-16:
On a related note: the sourceforge online code viewer has driven me up the wall enough times now that I've just cloned the RPi.GPIO repo [here](https://github.com/waveform80/rpi-gpio/) for the simple convenience of *viewing the damned code*. I am actually dreading submitting a pull request...

waveform80, 2016-02-08:
Haven't hit this since 0.6.1 got released; let's say it's fixed :)

closed 2016-02-08 by waveform80

sielay, 2020-05-23:
I noticed it back on 0.7.0 on Raspberry PI zero

Debian: 10.4
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

Linux raspberrypi 4.19.97+ #1294 Thu Jan 30 13:10:54 GMT 2020 armv6l GNU/Linux
processor	: 0
model name	: ARMv6-compatible processor rev 7 (v6l)
BogoMIPS	: 697.95
Features	: half thumb fastmult vfp edsp java tls
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xb76
CPU revision	: 7

Hardware	: BCM2835
Revision	: 900093
Serial		: 00000000793d6651
Model		: Raspberry Pi Zero Rev 1.3

Python 3.7.3

Code
```
import RPi.GPIO as GPIO

def setup():
  GPIO.setmode(GPIO.BCM)
  GPIO.setup(25, GPIO.IN, pull_up_down=GPIO.PUD_DOWN) # irc

def destroy():
  GPIO.cleanup()

def wait(pinNo, retry = 2):
  try:
    GPIO.wait_for_edge(pinNo, GPIO.FALLING)
  except RuntimeError as err:
    if (retry > 0):
      wait(pinNo, retry - 1)
      pass
    else:
      print("Runtime error: {0}".format(err))
      pass

setup()
wait(25)
destroy()
```

commit 2d161f56 (2015-10-12): Fix #50

This isn't a full fix but I can't seem to reproduce the issue in #50 at
the moment. So for now this just ensures that exceptions in the
constructors get cleaned up properly (so they don't block future
construction attempts against the same pin)
commit 8cfe8152 (2015-10-12): Merge pull request #60 from waveform80/safe-fail-init

Fix #50