gpiozero/gpiozero #51: Add circuit diagrams to docs

bennuttall, 2015-10-09:
I have some @samalder diagrams I can add, others will need creating especially (initially with fritzing is fine but Sam's are much better).

bennuttall, 2015-10-11:
First few have been added!

bennuttall, 2016-02-10:
Can we list missing diagrams here? (fritzling diagrams for Sam to redraw would be ace)

waveform80, 2016-02-10:
I've been leaving IMAGE TBD comments in docs/recipes.rst for "images to be done"; when I've finished a bit more code I'll see if I can whip up some fritzing diagrams for them.

waveform80, 2016-02-13:
Here's some more diagrams - do check I've got 'em right! I've include Fritzing diagram source for ease of modification in case I've got anything wrong.

For anyone that wants to modify these: work on the Fritzing diagram, save it, then export as SVG. Then run "make doc" in an appropriately configured venv (`make develop`). This'll use Inkscape to generate the PNG and PDF output (because it does this much better than Fritzing). Commit the Fritzing source, and all the image outputs (SVG, PNG, PDF); the reason for all the formats is that the SVG is the "canonical image" but Sphinx will use PNG for web output and PDF/EPS for PDF/Epub output (this is the reason the reST source uses `.*` extensions for images; it lets the Sphinx pick the most appropriate image format for the requested output).

bennuttall, 2019-02-10:
Just a suggestion to add to this:

I think we should document wiring diagrams separately from docstrings, on their own separate docs page, rather than sometimes in a docstring. Every device class (except HATs and things) should have a wiring diagram with useful notes (e.g. pull-up/pull-down) included. Some recipes have wiring diagrams, and I guess that's fine when multiple devices are used.

@waveform80 @lurch any thoughts? If agreed, I'll get started on it after the v1.5 release, for inclusion in the v1.6 release.

lurch, 2019-02-10:
...presumably with the class-docs linking to the wiring-docs (where appropriate), and the wiring-docs linking back to the class-docs?

bennuttall, 2019-02-10:
Exactly