gpiozero/gpiozero #53: Using blink as primitive

cymplecy, 2015-10-09:
I think making a LED blink is the 1st real program step in physical programming and having a command that does it is over-simplfying things too far.

Simplification should be used for complicated things like driving Neopixels/H-bridge motors etc

I think its also bit  debatable whether to use Led() straight out of the box just on simple things
I'd prefer a
LEDA = gpio17
while true:
  LedA(On)
  Wait(1)
  LedA(Off)
  wait(1)

type approach

bennuttall, 2015-10-09:
The first example in [recipes](https://github.com/RPi-Distro/python-gpiozero/blob/master/docs/recipes.md) in the docs:

> Turn an LED on and off repeatedly:

```python
from gpiozero import LED
from time import sleep

red = LED(2)

while True:
    red.on()
    sleep(1)
    red.off()
    sleep(1)
```

> Alternatively:

```python
from gpiozero import LED

red = LED(2)

red.blink()
```

So as you can see, I completely agree with you that the `while` loop version is a great starting point introducing the concepts.

However, `blink` gives you the ability to blink an LED in the background, which is a very handy feature. It means you can set it blinking, then do some other stuff without being held up by your blinking loop.

GPIO Zero isn't meant to dumb things down, just make it easier to get going. It's about being able to tell it what you want to do, not how.

cymplecy, 2015-10-09:
Well - I've never needed a threaded blinking LED before in a project myself  :)
But why have a LED class thingy - surely THAT is dumbing down :)
Surely the uses puts a LED on a pin - then says they want to control it.
Are you going to have a Bulb class, a solenoid class, anything else that goes on and off class ? :)

having a LED class makes sense for AddOn boards where the user doesn't know what pin the red led is connected to but its not right for plain pin control

IMO :)

bennuttall, 2015-10-09:
GPIO Zero is a library for everyday components. I started by implementing everything in the CamJam kits. I'll implement anything I think is useful. Anything not covered by these classes, you can use the generic `InputDevice` and `OutputDevice` classes, or just fall back to the brilliant `RPi.GPIO` library.

Class interfaces give context to code. You create an LED object, you expect to be able to turn it on, turn it off, etc. You create a Dog object, you expect it to bark.

closed 2015-10-09 by bennuttall

waveform80, 2015-10-09:
Sure we could have a Bulb class. It's definition will look like this:

```python
Bulb = LED
```

Funnily enough, that's more or less what the LED and Buzzer classes look like now (they're both empty subclasses of `DigitalOutputDevice` bar the explanatory doc-string).

cymplecy, 2015-10-10:
Your thinking like a pure software person.

But GPIOZero is a software interface to hardware. :)

waveform80, 2015-10-10:
Well, I am pretty much a pure software person, so I can't really argue there (though I'm not sure how a hardware perspective would change my view)! Still, I would object to removing stuff on the grounds given:

1. You've never needed a threaded blinking LED on a project before. That's fine, but it certainly doesn't mean that no-one's ever needed (or ever will need) a threaded blinking LED in a project (I can think of several cases off the top of my head: e.g. a police siren demo where red and blue LEDs flash while we play a nee-naw sound, and possibly drive towards something).

2. It's dumbing stuff down too far. As Ben's pointed out, just because something's there doesn't mean people have to use it (e.g. in lessons one can leave out `blink` until later). For that matter, from my experience with `picamera` and its users, just because something's there (and thoroughly documented) doesn't mean anybody actually notices (one could make similar arguments for `capture_continuous` and `capture_sequence` in picamera but it's rare I seem them used in practice - most people just use `sleep` and `capture` for stuff like timelapses).

   I'd also argue that the vast majority of beginners have no clue how to handle background threads, and it's probably not something we want to get into (in lesson situations) until much later. Yet the functionality is (or may be) useful in certain projects, and thus it's beneficial to expose a simple (limited) API to provide the functionality.

3. It'll lead to a proliferation of unnecessary classes. Again, I can think of a case where having otherwise redundant class aliases does serve a purpose. Consider the case of people using non-descriptive variable names in their script. Can you tell from the following script what's connected to pin 17?

   ```python
   foo = DigitalOutputDevice(17)
   ```

   We can tell it's an output device (so an LED, or a buzzer maybe?) but the following script is blindingly obvious:

   ```python
   foo = LED(17)
   ```

   So by encouraging use of these aliases we encourage nice clear code even in the event that people decide to use non-descriptive identifiers (it doesn't force people to use them instead of `DigitalOutputDevice`, but who wants to type all that?).

4. There is an argument to be made that with all those class aliases floating around doing something like `from gpiozero import *` pollutes the namespace excessively. But then this is only typical practice on the command line and is generally discouraged in scripts (in other words, I've got little sympathy for anyone that gets clobbered by using `import *`)

In other words, I can't say I'm convinced at the moment :)