gpiozero/gpiozero #56: Adding other helpful functions to GPIOZero

cymplecy, 2015-10-11:
If your going down the path of blinkin LEDs threading, then maybe worth considering other often needed functions.

Ultrasonic is 1st that springs to mind.

Motor wheel encoder counting is another

GetIP would be a non-gpio thingy  worth doing?

bennuttall, 2015-10-11:
Thanks for the suggestions. Could you expand?

Do you mean an ultrasonic sensor? That's definitely the sort of thing I want to add. If you can point me at a decent one I'll get hold of one and implement it.

I'm not sure what you mean by motor wheel encoder?

GetIP as in letting the user know what the Pi's IP address is without attaching a monitor? How would it convey the information?

cymplecy, 2015-10-11:
The standard HC - SR04

Some motors have quadrature rotation encoders to give pulses as they turn so that you can count the turns
https://shop.pimoroni.com/products/optical-encoder-pair-kit-for-micro-metal-gearmotors-3-3v?utm_medium=cpc&utm_source=googlepla&variant=798892261&gclid=Cj0KEQjwkeiwBRCzmo-wiKL49pEBEiQAhvGKYTMRoIXFhhUTBMFw6z0WhRr0nkYVfXptTUVAzNrT-C4aAk_Y8P8HAQ

GetIP instead of having to parse ifconfig - just simple active IP address (preferenece given to interface currently connected to LAN as opposed to non-active static assigned one)

bennuttall, 2015-10-12:
> The standard HC - SR04

Thanks Simon. I'll see if I can get hold of an ultrasonic sensor and add it to the list.

> Some motors have quadrature rotation encoders to give pulses as they turn so that you can count the turns

I'll also see if I can get hold of one of these here fancy motors. Can you point me in the direction of some code that does what you'd expect it to do?

> GetIP instead of having to parse ifconfig - just simple active IP address

You just mean a function to return the IP address? I'll look into it.

The easiest way to get your IP address on the command line is `hostname -I` (that's capital i) rather than using `ifconfig`. I bet there's a way of getting it quite easily in Python.

cymplecy, 2015-10-12:
em - I have some code to do the encoders in ScratchGPIO but its down to my usual standard but I think MartinH has done some good work in this area 

Re IP - yes - just simply return IP address

closed 2015-10-12 by cymplecy