gpiozero/gpiozero #57: Blink does not work for PWMOutputDevice

bennuttall, 2015-10-11:
When you try to blink a `PWMOutputDevice`, it flickers on and off quickly once, then stays off apart from random short flashes, even after turning off!

It may be caused by `blink` definition being inherited from `DigitalOutputDevice`, and it calling `super`'s `on` and `off`, but I'm not sure.

bennuttall, 2015-10-12:
Ok it was indeed caused by it calling `super`'s methods. I have a solution that fixes it, but I'm not sure this is the right way to go. Will try a couple of variations and send a PR for you to look at @waveform80.

bennuttall, 2015-10-12:
Ok PR created. I believe that commit fixes `blink` for `PWMOutputDevice`.

Explanation:

- `OutputDevice`'s `on()` method calls `GPIO.output()`...
- `DigitalOutputDevice`'s `on()` method calls `self._stop_blink()` method followed by `super.on()`
- `DigitalOutputDevice`'s `_blink_led()` method calls `super`'s `on()` method to stop it killing its own blink thread
- `PWMOutputDevice`'s `on()` method runs `self.value = 1` rather than an inherited `on()` method, which means that when the inherited `_blink_led()` method is called, it ends up running `super.on()` which cascades back to `GPIO.output`

So, rather than `_blink_led()` calling `super.on()`, I created `_on()` methods on `DigitalOutputDevice` and `PWMOutputDevice` which define the behaviour of turning the LED on without stopping the blink thread. `blink_led()` can now call `self._on()` whereas the exposed version `self.on()` both kills the blink thread and runs the appropriate turning on action.

(The same applies to `off`, of course)

@waveform80 Would you do it another way, or will this do?

waveform80, 2015-10-12:
It's a decent way, but I might push the idea a bit further partly for consistency and partly for future proofing. At the moment GPIODevice defines a `_read` method which is occasionally overridden for reading the value of a device. I'd be tempted, instead of adding `_on` and `_off` to DigitalOutputDevice, to add a `_write` method to GPIOOutputDevice which simply accepts a value and in the base class this is just cast to a bool and output with `GPIO.output` but is overridden in PWMOutputDevice to control stuff with `ChangeDutyCycle`.

This has a certain symmetry with `_read`, and extends the ability to change the meaning of "on" and "off" for any class of output devices. I'll have a play with this in a mo and see if it looks any cleaner.

waveform80, 2015-10-12:
Oooh - yeah, I think I've got something that's quite a bit nicer. It also makes `is_active` a lot more consistent (value of 0.0 is off, anything greater is on) and all the `DigitalOutputDevice` parent class methods "just work" now without any overriding (including blink, toggle, etc.)

bennuttall, 2015-10-12:
Ok sure - go ahead.

waveform80, 2015-10-12:
Okay, there's another PR which fixes things in a slightly different way. I think overall it's a bit nicer because it gives us a few more options down the line for PWM stuff, though I have removed min_pwm and max_pwm for the moment. They'll probably get added back in when we need them, but I need to think about how they interact with "value" and "is_active".

In particular, I have a feeling value should always be limited to 0 and 1 where 0 is off, 1 is on, and everything in between is a duty cycle for PWM related things. The reason is that this plays nicely into re-using all the base class methods (which can then safely assume writing 0 always means switching, and likewise 1 for on). So that would imply that if we specified a minimum and maximum duty cycle we should scale value such that 0 sets the minimum duty cycle (whatever that may be) and 1 the maximum. But anyway, I'm not 100% sure about this so I figured I'd just rip that stuff out for now and tackle it when we actually need to.

bennuttall, 2015-10-12:
Ok cool

commit bb6523a9 (2015-10-12): Fix #57

This refactors the GPIOOutputDevice to have an internal `_write` method
similar to the `_read` method used to grab the value of a GPIO device.
This is used for simple 1s and 0s until we get to PWMOutputDevice which
replaces the value with the PWM device's duty cycle level. As a result,
all the DigitalOutputDevice base methods (including blink) "just work".

This commit also, for the moment, removes min_pwm and max_pwm because
I'm not 100% certain how they should interact with the notion of
is_active at the moment. They'll probably get added back in at some
point but I need a bit more  [... cut]
commit f6c69e80 (2015-10-12): Merge pull request #61 from waveform80/output-read-write

Fix #57
closed 2015-10-12 by waveform80