gpiozero/gpiozero #58: Should LED extend PWMOutputDevice instead of DigitalOutputDevice?

bennuttall, 2015-10-11:
Making LED a PWM device by default would enable a brightness setting (if we aliased `.value` as `.brightness`):

```python
>>> led = LED(2)
>>> led.on()
>>> led.off()
>>> led.brightness = 0.5
```

Any reason not to? The alternative is leave it as it is, and people can manually create the led as a `PWMOutputDevice` object.

We would need to ensure #57 is fixed first.

bennuttall, 2015-10-14:
#57 is fixed

bennuttall, 2015-10-17:
@waveform80 what's the decision?

`LED` extends `PWMOutputDevice` or `LED` stays and we introduce `PWMLED`?

I think we said the latter.

waveform80, 2015-10-17:
Latter sounds good to me - I'm actually fine either way, but I have a tiny preference for having something brutally simple for LED (I'm just a nervous about RPi.GPIO's PWM implementation)

bennuttall, 2015-10-17:
Ok sure - I'll throw it in now.

commit 3230f3ae (2015-10-17): Add PWMLED, close #58
closed 2015-10-17 by bennuttall