gpiozero/gpiozero #64: PIR Motionsensor seems laggy and a bit unreliable

raspitv, 2015-10-13:
I guess it's something to do with the moving average way that it is being read, but I've tried two different PIR sensors with the same results, just using this code...

from gpiozero import LED, MotionSensor
from time import sleep
pir = MotionSensor(21)
light = LED(26)
pir.when_motion = light.on
pir.when_no_motion = light.off
sleep(30)

Using a 10k pull-down on the PIR output helps quite a bit, but the response is still rather laggy.

bennuttall, 2015-10-13:
I've not had a great deal of success with the PIR either. I've seen people tend to use a plastic cup to narrow the field of "vision" to an isolated area.

Worth noting the resistor helps - we could add that to the docs.

Any ideas @waveform80?

raspitv, 2015-10-13:
Actually with a 10k pull-down on the PIR output and this code...

from gpiozero import LED, MotionSensor
from time import sleep
pir = MotionSensor(21)
light = LED(25)
while True:
    pir.when_motion = light.off
    pir.when_no_motion = light.on
    sleep(0.1)

...it's working quite nicely now :)

bennuttall, 2015-10-13:
Really? That's odd.

waveform80, 2015-10-13:
I think the queue size on MotionSensor is probably rather too long. With the sensitivity set correctly the rising response of the PIR sensor (the one from the CamJam 2 kit) is actually rather good and the queue just makes it lag. The only issue is that the falling response, even with the timer pot set to the minimum, is still vast (a good second or so) which tends to make people think it's dodgy (it's not, it's just got a really long fall time).

The queue will still be really useful for anybody dealing with a PIR sensor which doesn't have logic on it with a long fall time, which is why I'm not in favour of removing it. But given the CamJam 2 kit's PIR sensor is likely to be the major use-case I'm tempted to change the default `queue_len` to 1. This might make your sensor appear more "responsive".

In the meantime, what's wrong with your code? I'm not sure on the first script - if the second is fine the first should be fine too. The second however, is repeatedly assigning handlers to the hooks which won't harm anything but is basically pointless. Have a go with something like this to see if the `queue_len` makes things better:

```python
from gpiozero import LED, MotionSensor
from signal import pause
pir = MotionSensor(21, queue_len=1)
light = LED(25)

pir.when_motion = light.off
pir.when_no_motion = light.on

pause()
```

waveform80, 2015-10-13:
Oh, and yeah - you might want to chop the bottom out of a paper coffee cup and stick it around the sensor as a "cone". We frequently have to do that in picademy to limit the field of view of the sensor otherwise it picks up movement from everywhere in the room.

raspitv, 2015-10-13:
Thanks. It's one of the small PIR sensors like in the CamJam kit, but from another source (tried two different ones). Everything turned down to minimum, with your latest code suggestion queue_len=1 it works well. It seems like the pull-down may not be needed either, but past experience shows it can reduce the likelihood of false triggering. Ta

bennuttall, 2015-10-13:
Great

waveform80, 2015-10-13:
Okay, I'll adjust the default `queue_len` down to 1 as evidently it's needed not just for the CamJam PIR sensors, but some others as well (and it certainly makes things more responsive for me too when testing). I'll also bung a note in the docs about increasing the `queue_len` if your sensor seems "jittery" (which would indicate no long fall time).

bennuttall, 2015-10-13:
That makes sense.

closed 2015-10-13 by waveform80

bennuttall, 2015-10-13:
Cool - I'll try this and see if it's effective for me. I find those things so unreliable! Would be good to just be a software config thing - they're great for projects.

commit 53f3a3c6 (2015-10-13): Fix #64

Make MotionSensor more responsive by effectively removing the averaging
queue. Also add note on how to smooth out "jittery" PIR sensors by
increasing the queue length back up to 5.
commit 612975e6 (2015-10-13): Merge pull request #67 from waveform80/response-pir

Fix #64