gpiozero/gpiozero #68: Setting individual colours for RGBLED via setters does not work

snake48, 2015-10-13:
Can be fixed by changing:

   @red.setter
    def red(self, value):
        self._red.value = self._validate(value)

to

   @red.setter
    def red(self, value):
        self._red._set_value(value)

and similarly for green and blue setters.

waveform80, 2015-10-14:
Yup - there's some bugs there. I'll get it patched up in a bit.

waveform80, 2015-10-14:
Hmm, actually there's a few more things that need fixing here. For starters, RGBLED needs a `close` method for proper clean-up (useful when re-wiring things at the command line) and secondly, PWMOutputDevice isn't cleaning stuff up properly in its close (PWM threads are left hanging around with nasty consequences). I'll fix the lot in a PR in a bit but it's going to be a bit more complex than I'd initially thought...

bennuttall, 2015-10-14:
Thanks Dave

closed 2015-10-14 by waveform80

waveform80, 2015-10-14:
Ooh - just realized this works really nicely with picamera's `Color` class now (because that uses RGB 0-1 representation internally). So you can do stuff like this:

```python
from gpiozero import RGBLED
from picamera import Color

led = RGBLED(12, 20, 16) # or whatever pins you've got stuff wired to
led.color = Color('#ffffff') # supports HTML color specs
led.color = Color('violet') # or CSS color names
led.color = Color(255, 127, 0) # or 0-255 bytes
led.color = Color(0.0, 0.5, 0.0) # or 0-1 floats
led.color = Color(hue=0, saturation=0.5, lightness=0.5) # or HSL
```

Now I'm tempted to make the getter return a picamera.Color too so you could do cool stuff like:

```python
led.color = led.color - Saturation(0.5) # partly desaturate the color presented
led.color += Hue(0.1) # rotate the hue around
```

But that'd add a dependency on picamera which is a bit silly considering. I really ought to finish off the Color class properly and split it out into its own package (it's not dependent on anything in picamera particularly)...

commit 250ba454 (2015-10-14): Fix #68

Actually, this isn't so much a fix as a re-working of RGBLED. This
renames the `rgb` property to `color` because that's what the property
really represents (combined with `picamera.Color` this lets you do stuff
like `rgb_led.color = Color('red')`), and simplifies the internals of
the class. It also adds the common `close` method and context manager
capabilities.

Finally, it corrects an error I hadn't noticed earlier where
PWMOutputDevice isn't properly cleaning up PWM threads on close.
commit 7dda0fe5 (2015-10-14): Merge pull request #72 from waveform80/fix-rgb-leds

Fix #68