gpiozero/gpiozero #69: Energenie

bennuttall, 2015-10-13:
It was suggested in #66 that we add an [Energenie](https://energenie4u.co.uk/index.phpcatalogue/product/ENER314) class.

I maintain the [energenie](https://pypi.python.org/pypi/energenie) module, which provides a simple interface to the Energenie Pi-mote. It's just a case of turning pins on and off in sequence.

Seeing as I'm the maintainer and it's not going to upset anyone, should we consider incorporating Energenie functionality into GPIO Zero?

I'd consider this low priority, would rather concentrate on other things before the 1.0 release, but wanted to start a thread now.

waveform80, 2015-10-25:
Sounds like the best excuse I've had yet to actually do something with my energenie plugs! I'll try and find some time to look at the existing module and what options we've got for integration in a consistent manner.

bennuttall, 2015-10-25:
The logic for working out which bits to flip in there, but I see our intended interface being very different.

The way it currently works is you pass in a switch number e.g. `switch_on(1)`. There was a PR to use `*args` to allow `switch_on(1, 2)` to switch on `1` and `2` simultaneously which I liked but I never got around to dealing with it.

I imagine the GPIO Zero way would be to provide `energenie.on()` and maybe `energenie.one.on()` through `energenie.four.on()` but would that make it harder to use?

waveform80, 2015-10-25:
Hmmm... Well, the GPIO Zero way is to provide an enumerable list of sub-devices (e.g. LEDBoard's leds property) but provide natural aliases for items within that when we know what they are (TrafficHat's red, amber, and green properties which are trivial aliases of leds[0], leds[1], and leds[2], inherited from the superclass).

In the case of Energenie, I'd certainly have an enumerable list of sub-devices (devices[0], switches[0], or whatever) because that allows you to easily address a device with an index number, but natural aliases would be nice too. I'm just wondering how to do that...

We could allow a constructor that takes `**kwargs` and uses the names to generate property aliases. So for example:

```python
e = Energenie(lamp=0, fridge=1)
e.lamp.on()
e.fridge.off()
# Which is equivalent to
e.devices[0].on()
e.devices[1].off()
```

There's a limit to the number of devices isn't there - 4 or something (I haven't got the box in front of me at the moment).

waveform80, 2015-10-25:
And then of course have the usual `on` and `off` methods on the class itself which control all devices within it.

bennuttall, 2015-10-25:
Oooooooh that's a nice idea.

There are two products: a single socket unit, and a four socket unit.

The problem is: the switches are physically labelled 1 to 4, so using 0 to 3 doesn't really make sense.

waveform80, 2015-10-25:
Oh - hang on, forget that. It's me trying to be too clever again, and winding up being stupid. It's over-complicated, inconsistent with the rest of the library and just plain weird. A better idea is just to follow what we've already done with another composite device which has arbitrary attachments: ADCs.

Look at the MCP3008 class. It doesn't represent a single MCP3008 chip: it represents a single *channel* of that chip. If you want to address multiple channels you create multiple instances (we haven't defined what happens if you double up on instances for a channel, but nevermind - we'll tackle that when someone hits it!). Anyway, the point is that this works really nicely. You could have any number of things attached to each channel of an ADC so just representing a single channel of it in an instance allows you to use variable names to alias things nicely:

```python
light_sensor = MCP3008(0, 0)
thermistor = MCP3008(0, 1)
print(thermistor.value)
```

Likewise, our Energenie class should represent a single channel of the Energenie adapter so the user can utilize variable names appropriately:

```python
lamp = Energenie(0)
fridge = Energenie(1)
lamp.on()
fridge.off()
```

This keeps the constructor consistent with most other classes (instead of introducing weirdness like **kwargs which can go horribly wrong if someone actually unpacks a dict into it and you start assuming all keys are actually valid identifiers) and is still consistent in behaviour with most other things in the library.

It does sacrifice the ability to turn everything on and off at once, but is that a sufficiently common use case to justify moving to a more complex class?

waveform80, 2015-10-25:
(oh, and if the channel names are typically 1-based instead of 0-based, then we should use that instead - no sense making things overcomplicated for people wanting to migrate - I just used 0-based in the examples above out of force of habit ... and come to think of it GPIO isn't 0-based either)

bennuttall, 2015-10-25:
That does make more sense!

However, I've a feeling the way it works is that the data you have to send the Energenie is coupled with the other sockets - in that you need to know the state of all four before sending a command to change the state. (It's been ages since I looked at it...)

waveform80, 2015-10-25:
That's fine - I can finesse that behind the scenes (have instantiation of the initial object trigger the construction of a shared background object that all further instantiations talk to, and have finalization of the last object destroy it - vaguely similar to the stuff we do with GPIO pins in GPIODevice's __init__ and close methods at the moment).

bennuttall, 2015-10-25:
Cool.

PR #239 (2016-04-01): Fix #140, fix #69, fix #185
This PR adds a software SPI implementation. Firstly this removes the absolute necessity for spidev (#140), which also means when it's not present things still work (effectively fixes #185), and also enables any four pins to be used for SPI devices (which don't require the hardware implementation).

The software implementation is simplistic but still supports clock polarity and phase, select-high, and variable bits per word. However it doesn't allow precise speeds to be implemented because it just wibbles the clock as fast as it can (which being pure Python isn't act [... cut]
commit 759a6a58 (2016-04-01): Fix #140, fix #69, fix #185

This PR adds a software SPI implementation. Firstly this removes the
absolute necessity for spidev (#140), which also means when it's not
present things still work (effectively fixes #185), and also enables any
four pins to be used for SPI devices (which don't require the hardware
implementation).

The software implementation is simplistic but still supports clock
polarity and phase, select-high, and variable bits per word. However it
doesn't allow precise speeds to be implemented because it just wibbles
the clock as fast as it can (which being pure Python isn't ac [... cut]
commit a7b7fc8d (2016-04-01): Merge pull request #239 from waveform80/energenie

Fix #140, fix #69, fix #185
closed 2016-04-01 by waveform80