gpiozero/gpiozero #74: led.is_on not led.is_active

bennuttall, 2015-10-17:
Someone suggested on the forums that the `LED` class's `is_active` property should be named `is_on`. I think it's a good idea and aligns with other aliases like `is_pressed` in `Button`.

@waveform80 can we move `_alias` from `input_devices.py` to `devices.py` and use it in both input and output devices?

waveform80, 2015-10-17:
Sure, sounds good to me

waveform80, 2015-10-17:
Huh, as part of this I figured it was about time I went and read up on `super()` properly in case there's a nicer way of aliasing properties than lambdas and string lookups (the current method we're using is basically a horrible hack).

The [official docs](https://docs.python.org/2/library/functions.html#super) are a bit vague on its use outside of methods. Raymond Hettinger's [post](https://rhettinger.wordpress.com/2011/05/26/super-considered-super/) sheds a bit of light on things but still doesn't go into its use in class definitions.

Anyway, after lots more googling and a fair amount of experimentation it turns out that `super` can be used in the class definition but only *after* it's finished (because until it's finished the class' name hasn't been bound to the definition so you can't refer to it in the `super` call). So, there's a couple of ways of doing this "properly" (i.e. without relying on lambdas and string lookups as we currently are ... eurgh). So, given the following definition, how do we create a class B which aliases the `value` property to `number`:

```python
class A(object):
    @property
    def value(self):
        return 1
```

The first method is to refer directly to the base class:

```python
class B(A):
    number = A.value
```

This works (in py 2 and 3), but it's icky because of the base-class reference (the very thing `super` is meant to get rid of). Why can't we just refer to `value` without the base-class qualifier? Because until B's definition is complete it doesn't inherit anything from A (that's effectively done at the *end* of the class definition). So ... method 2 is define the aliases once the definition is finished:

```python
class B(A):
    pass

B.number = B.value
```

This works nicely. It's a bit icky (referencing the class name again and again) but there's no base references, no lambdas, and no string lookups. So, I figure it's better than the current method. I'll do a PR for all this in a bit.

waveform80, 2015-10-17:
BTW, I think `is_lit` is probably a bit more specific than `is_on` ("on" is a bit generic)

commit 65832232 (2015-10-17): Fix #74

Add more property aliases and do them properly (no more lambdas and
string lookups) which means we can remove `_alias`. This commit also
defines `__slots__` for all classes which should prevent assignation of
invalid attributes with an AttributeError (more friendly than silently
doing the wrong thing). Finally, it cleans up all the property defs to
use Ben's preferred decorator style.
commit fec4bcc8 (2015-10-17): Merge pull request #77 from waveform80/moar-aliases

Fix #74
closed 2015-10-17 by waveform80