gpiozero/gpiozero #75: Change "Blink" to "Buzz" for buzzers.

ryanteck, 2015-10-17:
While not critical when writing my tutorial for TrafficHAT Involving buzzing the buzzer it felt a bit weird explaining to blink the buzzer. Where as being able to buzz the buzzer seems much more friendly.

bennuttall, 2015-10-17:
I agree. We've discussed this before. I just don't think `buzz` is the right word to use (I would assume `buzz` would turn the buzzer on), and I can't think of a better one.

waveform80, 2015-10-17:
Hmm, just pondering this and I wonder if this is a solution:

`pulse` is a pretty generic term that can be applied to just about any device. The only issue with it is that in the case of LEDs, people may expect it to do something similar to the `pulse` method of the Explorer HAT (which "smoothly" blinks the LEDs via PWM). So...

Given how generic the term `pulse` is, we stick it in as a method on the base OutputDevice class with usual two `on_time`, `off_time` parameters. This is then inherited by LED and we can simply alias it as `blink` (partly to maintain compatibility and partly because it's a nice obvious alias and we're using those in other places, so why not). We could also alias it to `buzz` in Buzzer if we wanted (BTW, has anyone tried PWM with a buzzer yet?) or maybe `siren`?

In PWMOutputDevice, we can then extend the prototype of `pulse` to include two additional parameters: `fade_in_time`, `fade_out_time` (maybe those aren't the best names ... erm ... `attack_time`, `decay_time`? Or maybe `ramp_on_time`, `ramp_off_time`? Well whatever, you get the idea). These default to 0.0 (so the method does the same as its ancestor by default) and can be called like its ancestor or can be called to do the nice "smooth" blinking in descendents like PWMLED (as in #58).

Thoughts?

ryanteck, 2015-10-17:
With PWM on a buzzer I don't find much of a use for it as it just makes it sound even more horrible. Instead a frequency change may work to change the sound made which then it would have a use?

Weather buzz is the correct word or not I don't know. 

Siren may be good for a fading up and down version of Pulse for LED.

bennuttall, 2015-10-18:
Actions:

1. Let's play around with PWM and try to find a use for it with the buzzer
2. Let's try to think of a good word to use as an alias for `blink` that suits the behaviour when used with a buzzer

bennuttall, 2015-11-12:
We might just have to go for a word like "beep"

ryanteck, 2015-11-12:
Beep sounds cool to me. Beep beep

*Thanks -*

*Ryan Walmsley*
*Ryanteck LTD.*
*Email: *Ryanteck@gmail.com
*Skype:* Ryanteck
Website:* Http://ryanteck.org.uk <Http://ryanteck.org.uk>*

waveform80, 2015-11-14:
`beep` sounds good to me; I'll get on it tomorrow

waveform80, 2015-11-15:
Had a play with PWM while doing this; it *sort of* changes the volume at low values, but the results are pretty horrid. I wouldn't feel bad just leaving it out for now.

bennuttall, 2015-11-15:
Interestingly, the guys developing the Python API for the BBC micro:bit have created a very cool music API which works when attaching a basic speaker or even the buzzer (like the camjam kit one). I asked how it worked and they said it's just calculated frequencies and duty cycles using PWM. I haven't looked at the code yet but I'd definitely like to see if we can borrow the idea from them. Not now - it can come after the release!

closed 2015-11-15 by waveform80

waveform80, 2015-11-15:
I could well believe that works with a completely reliably timed PWM implementation, but I'm not sure the software based one in RPi.GPIO is up to that (RPIO's DMA one probably is). My main reason for scepticism is that PWM with an LED occasionally produces "bright flashes" which I suspect is where the PWM thread gets interrupted by the OS to run something else, so its timing is disrupted. With sound that would produce audible "blips". Still, it's a fascinating idea and we should certainly explore it!

commit 4b7a923a (2015-11-15): Merge pull request #106 from waveform80/beep-beep

Fix #75
commit f69b6195 (2015-11-15): Fix #75

Alias blink to beep for Buzzer