gpiozero/gpiozero #79: Change Motor's "back" init arg to "reverse"

bennuttall, 2015-10-18:


bennuttall, 2015-10-18:
Should the `backwards` method be called `reverse` too? Same would apply to the `Robot` interface.

waveform80, 2015-10-19:
Good question. I'd suggest changing it to "backward" just to match the corresponding method name (so if people remember the method name they implicitly remember the parameter name too).

On a similar note, now we've got `values` in the base GPIODevice class, I'm pondering how to add something similar to Motor and then Robot. There's two ways of going about this:

1. Expose the underlying forward and backward objects so that people can tie things directly to `motor.forward_device.source` (or something similar).

2. Expose a `source` (and `values`) for Motor directly where, say, +1.0 means it's going forward full speed, and -1.0 means going backward full speed.

The latter appeals to me in some way, but then there's a question of what you do with a two motor robot class. After all, if Motor has values, shouldn't Robot? So, what does turning left equate to? Turning right? Going forward? Spinning round in circles? All of a sudden it's difficult to map to a single ... value ... huh ... compound values ... tuples. Of course, why didn't I think of that until I wrote all this out. Okay, never mind, I know where to go with this now.

bennuttall, 2015-10-19:
>  I'd suggest changing it to "backward" just to match the corresponding method name

I'm proposing that they both be called `reverse`. Thoughts?

> Expose the underlying forward and backward objects so that people can tie things directly to `motor.forward_device.source` (or something similar).

Yes please!

> Expose a `source` (and `values`) for Motor directly where, say, +1.0 means it's going forward full speed, and -1.0 means going backward full speed.

Yes please!

I saw an example like this with an early Flotilla prototype. It used two slider pots to control the left and right motors respectively: high pot value = forwards; low pot value = backwards; mid point pot value = still. Very neat! Though requires mapping `1` to `forward(1)`; `0` to `backward(1)`; `0.5` to `stop()` and everything in between accordingly. Alternatively there are cases you'd use more or fewer pots...

bennuttall, 2015-10-22:
**Decision: leave as `backward()`**

However, we should also update the init args of `Motor` to `forward` and `backward`.

commit 8bdbac26 (2015-10-22): Merge pull request #86 from waveform80/moar-sources

Fix #76, fix #79
commit fa0a1b3c (2015-10-22): Fix #76, fix #79

This finishes off implementing values and source for all (current)
classes in gpiozero. I'm afraid things get rather complex in this
commit. For starters, we've now got quite a few "aggregate" classes
which necessarily don't descend from GPIODevice. To implement values and
source on these I could either repeat a helluva lot of code or ... turn
to mixin classes. Yeah, it's multiple inheritance time, baby!

Unfortunately multiple inheritance doesn't work with __slots__ but we
really ought to keep functionality that they provide us (raise
AttributeError when an unknown attribute [... cut]
closed 2015-10-22 by waveform80