gpiozero/gpiozero #87: Add test suite

bennuttall, 2015-10-23:


icarter09, 2015-12-01:
Is there any certain criteria that we wanting in this test suite? Take class LED for example, would we want to test that an LED must be instantiated using the correct arguments and that each method performs correctly?

waveform80, 2015-12-01:
That's basically the size of it - I'm holding back on doing this just yet because at the moment I'm re-writing some of the guts of the library to permit future expansion using I2C based IO expanders and similar things (basically I'm teaching the library a generic "pin" concept and then building a GPIO-specific sort of pin on top of that as the default pin type). But as a consequence of this quite a lot of the low level guts (devices.py in particular) is changing and that'll impinge rather heavily on any test suite (I might write the basis of a test suite while I'm at it anyway).

icarter09, 2015-12-02:
Sounds good. Do you need any help on writing up a test suite for the rewrite you are doing?

carlosperate, 2016-01-15:
Hi guys,

I was wondering what the general plan for testing is for the module with respect of the hardware.

The reason I ask is because I was toying with this idea in my head to maybe monkey patch all of Rpi.GPIO and there could be some code re-utilisation there if that is something you guys were thinking about implementing as well.

waveform80, 2016-02-11:
Okay, here's what's going on with testing:

I'll be pushing some stuff tonight which will form the beginnings of the test suite. Now that the pin implementations are split out there'll be two sides to the test suite much like the library itself: tests for devices (InputDevice, OutputDevice, PWMOutputDevice, etc.) and tests for pin implementations (RPiGPIOPin, RPIOPin, NativePin, etc.).

To support the device tests I've whipped up a MockPin implementation which records its state transitions and provides some rudimentary assert methods to test that the pin passed through expected states (at expected times if required, for testing things like blink()). This means that device tests can be entirely hardware independent and when things are a little more complete I'll put together a Travis config for that stuff.

Pin tests are going to be rather more complex because what we really want from a pin implementation is to test that it actually drives the hardware properly. However, we don't really need to test every pin (on the assumption that any vaguely sane library won't be doing different things for different pins). I can probably come up with a reasonably simple test setup, e.g. wire one GPIO pin to another and then do things like:

* Set a pin to input, set another to output and use the input pin to check it drives high and low as requested.
* Set the output pin to input then set pull up on the pin and use the other pin to check the pull up works.
* Set the output pin up for PWM, set a 50% duty cycle, sample a hundred reads off the input pin and accept the result if it's within some tolerance of 50% (repeat for various duty cycles)
* And so on...

Given that the idea of the Pin interface is to provide a common interface to underlying GPIO libraries we should be able to make a single test suite that covers all implementations (at which point pytest's awesome fixtures can be used to fire it off for each implementation).

I've got a couple of Pis here which form my picamera test bed (a pi 1B and a pi 2B which both have cameras permanently attached). They don't have anything on their GPIO pins but I could easily wire them up for such a test bed and then set up some sort of GitHub webhook to fire up the test suite when commits/PRs appear on the main repo. For the Travis config we can simply set the pin tests to xfail or something.

Anyway, that's the plan. The MockPin implementation is done and I'm working through output devices right now to give people an example of the sort of tests we're looking for. Coverage configuration is included and once I've merged this branch you should be able to do the following to try out the test suite:

* Clone the main repo
* Set up a venv to work in
* Within the venv run `make develop` to install all development/doc/test dependencies
* Within the venv run `make test` to run the test suite with coverage (this can be done on any machine, not just a Pi but when pin tests are added they'll only run on a Pi)

Unfortunately what with Picademy Manchester starting next week, I'm not going to have a lot of time to work on test cases (or much else!) ... so if anyone wants to extend our coverage stats, PRs will be very welcome!

waveform80, 2016-02-11:
Hmm, haven't got access to add set up the repo in Travis: @bennuttall - can you do the necessary bits? I've included a travis config in the PR so as soon as it's merged it should start running the (few) device tests I've written.

bennuttall, 2016-02-11:
Looking into it!

waveform80, 2016-02-11:
Thanks - it's working :)

bennuttall, 2016-02-11:
So it is!

waveform80, 2016-02-11:
Woohoo - our first failures :)

waveform80, 2016-02-11:
I'll fix 'em and update the PR but then I think this is probably ready to go in.

bennuttall, 2016-02-11:
Great!

waveform80, 2016-02-11:
I'll need to rebase the ultrasonics PR after that but that'll only take me a few minutes (both PRs introduce a compat module for backported stuff)

waveform80, 2016-02-11:
Okay, one test failed because of a tiny timing slip, and that was almost certainly due to PyPy taking ages to execute it. Everything else passes though, and there's plenty for people to pour over here so I'm going to go ahead and merge this then rebase ultrasonics. For anyone that wants to work on tests see above for installation instructions (it's easy with a venv) and check out tests/test_outputs.py - that should give you an idea how to use the mock pin implementation and what tests are generally expected to look like (basically simple functions with names that begin with "test_" containing a few assertions).

bennuttall, 2016-02-11:
Awesome, thanks

waveform80, 2016-02-11:
Right, there we go: first batch of tests are now passing nicely. Ahhh ... I feel more relaxed already! (funny how having a test suite does that :)

bennuttall, 2016-02-11:
Great :)

waveform80, 2016-04-05:
@lurch - sorry I'm being terribly lazy! Can you remember where we discussed testing PWM with statistical measurement - I vaguely recall you had some fantastically good results with this but I haven't found the thread in a few minutes of searching and wondered if you could recall off the top of your head where they were?

lurch, 2016-04-05:
Eventually managed to track it down... #220 ;-)

waveform80, 2016-04-05:
How the hell did I miss that? I was searching for "PWM and test" and couldn't find it ... oh, it's a PR ... more `self.kick()` calls required ...

waveform80, 2016-04-10:
Okay, I'm happy enough with coverage for now (on an actual pi with the real pins tests it's over 70%) that I'll consider this done for 1.2. However, we're not near 90% coverage yet, and there's plenty more to do so I'm going to re-target this for 1.3 to remind us to work on it more then.

bennuttall, 2016-05-29:
How happy are you with the test suite now @waveform80?

waveform80, 2016-06-04:
I'm happy enough with the test suite for 1.3's release. But I wouldn't close this ticket - just bump it to 1.4. We're still nowhere near 90% coverage (largely due to real-pin library tests) and I've got to write a proper test suite for all the SPI stuff.

Incidentally, huge thanks to @lurch for his numerous contributions edging our coverage ever upwards!

bennuttall, 2019-02-01:
"Yeah, that's all done" - @waveform80 

#725

closed 2019-02-01 by bennuttall