gpiozero/gpiozero #90: Optional PWM interface to boards?

bennuttall, 2015-10-25:
I can create an instance of a `TrafficHat`, but then I have access only to `LED` objects, meaning I can't dim or pulse the LEDs.

How about adding a PWM option on init?

```python
th = TrafficHat(pwm=True)
```

Then the LEDs would be created as `PWMLED` objects.

waveform80, 2015-10-25:
Nice!

bennuttall, 2015-10-25:
Cool, I'll see what I can do!

bennuttall, 2015-10-25:
Oh, you self-assigned. Feel free - otherwise I'll happily tackle it.

waveform80, 2015-10-25:
Go for it if you've got the time - I'm probably going to be rather busy with HackMcr Jr tomorrow and Tuesday (might have some time in the evenings).

bennuttall, 2015-10-25:
Ah, yes. Good luck with it!

bennuttall, 2015-10-25:
Implemented in branch `fix-90`. Seems to work. Quite a handy option.

In `LEDBoard` I do it this way:

```python
def __init__(self, *pins, pwm=False):
    super(LEDBoard, self).__init__()
    LEDClass = PWMLED if pwm else LED
    self._leds = tuple(LEDClass(pin) for pin in pins)
```

And added it to other boards.

With `TrafficHat` I can set `th.lights.red.value = 0.5` but can't seem to set `th.lights.value = (1, 0.5, 0.1)`?

@waveform80 what do you think?

waveform80, 2015-10-26:
Huh, that's weird - I'll look into it as a superficial glance at the code suggests that ought to work...

waveform80, 2015-10-26:
Oh, one little problem - that syntax you've used in `LEDBoard.__init__` is Python 3 specific (in Python 2 you can't have anything by `**kwargs` after `*args` because Python 2 had no concept of keyword-only args). I'll tweak it so it's compatible with both but I'm afraid it's going to look a little ugly.

waveform80, 2015-10-26:
Ahh ... I'm beginning to remember why I started avoiding the decorator syntax for properties. This is a [long-standing issue](http://bugs.python.org/issue14965) (although it's seen more as a feature by many on python-dev) with `super()`. There's two ways to work around it...

One: stick with decorator syntax and use the following body:

```python
@property
def value(self):
    return TrafficLightTUple(*super(TrafficLights, self).value)

@value.setter
def value(self, value):
    super(TrafficLights, self.__class__).value.fset(value)
```

Two: switch to method syntax. With this syntax you can just call the super-class' property methods directly (because they have different names):

```python
def _get_value(self):
    return TrafficLightTuple(*super(TrafficLights, self)._get_value())

def _set_value(self, value):
    super(TrafficLights, self)._set_value(value)

value = property(_get_value, _set_value)
```

I've gone with the first just to maintain consistency in the use of the decorator syntax.

PR #95 (2015-10-27): Fix #90
This is a combination of Ben's original commit to fix #90 with a couple of tweaks to ensure Python 2 compatibility and get the super-call in the value setter to work properly. Tested with both Py 2 and 3, seems to work nicely.
commit aa542f2f (2015-10-27): Merge pull request #95 from waveform80/fix-90

Fix #90
closed 2015-10-27 by waveform80