gpiozero/gpiozero #91: Add toggle to Motor and Robot

bennuttall, 2015-10-25:
`motor.toggle()` would reverse the direction of the motor if moving (forward <=> backward).

I guess it would run `motor.forward()` if the motor is not in motion.

`robot.toggle()` would reverse the direction of the robot if moving (forward <=> backward; left <=> right).

I guess it would run `robot.forward()` if the robot is not in motion.

waveform80, 2015-10-25:
I'm not so sure about the `robot.toggle()` suggestion. To me, the concept of the `toggle()` method (which is so far true of all our implementations of it) is that running it twice (or any even number of times) should result in the object being left in the original state. However, under the specification given, if the robot is stopped, running it once starts it moving forward, then again moves it backward. No method is given for returning it to the original state via `toggle()`.

Frankly, for things like motors I'm not sure `toggle()` makes much sense (we're not really switching between two distinct states). I do like explorer HAT's `reverse()` though. I could see that having a sensible definition in the case of motor via negation:

```python
self.value = -self.value
```

In other words, it we're running at 1 (full speed forward), run at -1 (full speed backward). If we're at -0.5 (half speed backward), change to 0.5 (half speed forward). And if we're at 0, remain stopped.

A similar definition can also be applied to robot:

```python
self._left.value = -self._left.value
self._right.value = -self._right.value
```

Thus, if the robot's running full speed forward, it'll run full speed backward. If it's turning left (left=-1, right=1) it'll turn right (left=1, right=-1), and so on. If it's stopped, it'll stay stopped which is why `toggle()` really isn't a good name for this - but `reverse()` makes good sense.

bennuttall, 2015-10-25:
Yeah I wasn't sure. I don't like inconsistent toggle behaviour either, so I guess we should drop it. Do we still have `toggle` for PWM devices, or did you remove it? That was also inconsistent.

`reverse` does make sense for `Motor` and `Robot`.

waveform80, 2015-10-25:
Good question - I'll just check...

waveform80, 2015-10-25:
Hmmm, PWMLED inherits toggle from DigitalOutputDevice which simply queries is_active (which in the case of PWMOutputDevice returns true if value > 0.0, i.e. if the device has any duty cycle at all) and calls on() or off() in response. So, in the case of a PWMLED, if the device's duty cycle was, say, 0.1, a call to `toggle()` turns it off, then a second call turns it all the way up to 1.0. Not sure I like that.

I'm tempted to come up with a more reversible transform like `1.0 - value` which would flip between 0 and 1 at the extremes but between 0.1 and 0.9 in between and would effectively do nothing at 0.5. Thoughts?

bennuttall, 2015-10-25:
I think that's better than the current behaviour. Go for it.

closed 2015-10-27 by waveform80

bennuttall, 2015-10-27:
Nice! Thanks

commit 0c8eff6a (2015-10-27): Merge pull request #96 from waveform80/pwm-toggle-motor-reverse

Fix #91
commit 2a1ed115 (2015-10-27): Fix #91

Override toggle() in PWMOutputDevice, and add reverse() to Motor and
Robot