gpiozero/gpiozero #92: MCP300X argument ordering?

bennuttall, 2015-10-25:
It occurred to me recently that you're more likely / more often to set the `channel` than the `device` in `MCP3008` initialisation. Therefore, should the argument order be reversed? i.e.

```python
a = MCP3008(1)  # equivalent to MCP3008(channel=1)
```

For:

- Easier to address quickly
- Least significant bit first makes sense (or does it?)

Against:

- You can use named arguments anyway e.g. `channel=1`
- Most significant bit first makes sense (or does it?)

waveform80, 2015-10-25:
Hmm ... good question! I'll have a think about that.

bennuttall, 2015-10-25:
Like with many of these questions and suggestions, I'm sitting on the fence. I think we've made some good decisions by talking it through in issues.

bennuttall, 2015-11-02:
Any thoughts?

waveform80, 2015-11-03:
Not yet - but I'm still recovering from picademy :) I'll get on this tomorrow!

lurch, 2015-11-11:
Whatever the outcome from this issue, it seems clear that the documentation for the MCP3008 needs tidying up - it talks about e.g. `wait_for_light()` ??

waveform80, 2015-11-14:
Well, I completely failed to get on this 11 days ago (according to GitHub). Sorry - been busy with picademy and software carpentry obligations. Tomorrow, honest!

bennuttall, 2015-11-15:
Ok decision is to go ahead and swap argument ordering

bennuttall, 2015-11-15:
@waveform80 did I miss anything?

bennuttall, 2015-11-15:
Technically the examples in the docs that use `MCP3008(channel=1)` are not wrong but could be changed to  `MCP3008(1)` but I think it's ok to leave it.

closed 2015-11-15 by bennuttall

waveform80, 2015-11-15:
Yeah, leave the docs as is - it makes things absolutely clear that you're talking about the channel and people can always figure out that it can be specified positionally.

commit 382d6e45 (2015-11-15): Reverse order of MCP300X arguments, close #92