gpiozero/gpiozero #94: Blink doesn't work for boards when background=False

bennuttall, 2015-10-25:
Is this fixable? Or should we drop `background` in these cases?

waveform80, 2015-10-26:
It's fixable but tricky - basically `background=False` in those cases actually has to launch a background thread anyway to get them all going at the same time, then wait for it to terminate. I'll deal with when I get a sec.

bennuttall, 2015-10-27:
Ok cool - thanks.

lurch, 2016-02-01:
Dunno if it's related to this, but if I construct an LEDBoard and call `board.blink()` then all the LEDs (I'm using 8 LEDs here) flash on and off at the same time. But if I call `board.blink()` a second time, while my LEDs are already blinking, then the LEDs change to visibly turning on and off one at a time? (but calling board.blink() a third or fourth time doesn't change the 'slow' behaviour)

On a slightly-related topic a) should LEDBoard have a `_stop_blink()` method too, and b) what's the reason for `DigitalOutputDevice._stop_blink` being 'internal' (i.e. it has a leading underscore) anyway? Is there any reason it couldn't be a 'public' method? I guess it's not immediately obvious (until you actually try it) that using `on()` or `off()` also stops the blinking.

Hmmm, and I wonder if it'd be worth adding an `initial_value` parameter to the `blink()` methods too, so that if you're using a long blink-time, you can choose whether to have it starting-on or starting-off?

waveform80, 2016-02-02:
Interesting point on `_stop_blink` being internal. I'd assumed there was no use case for it to be public, so why bloat the API? But now I think about it, there's one use case: start a blink and then stop it leaving the device in whatever state the blink was in (at the moment you can only stop and force the device on or off). Is that sufficiently useful to justify it being public?

For initial value, at the moment blink always starts on (in the case of LED) and fading-to-on (in the case of PWMLED), but there's certainly a case to be made for starting off (if the LED's already on). Obviously the default should be start on (if only for backward compatibility), but it's a decent idea. Any chance you could open a ticket for these?

lurch, 2016-02-03:
> Is that sufficiently useful to justify it being public?

Well it exposes extra functionality, and I think it makes a nice counterpart to the `blink` method.

> there's certainly a case to be made for starting off (if the LED's already on).

Or maybe you e.g. want an off-time of 10 seconds, and an on-time of 0.5 seconds, but you don't want an initial flash when you first run your program.

> Any chance you could open a ticket for these?

Will do...

PR #161 (2016-02-07): Add a blink thread to LEDBoard
Fixes #94
commit 3ed1ab84 (2016-02-07): Add a blink thread to LEDBoard
Fixes #94
closed 2016-02-07 by waveform80