gpiozero/gpiozero #98: Question: How do you generate beautiful diagram circuits ?

jmrenouard, 2015-11-01:
There is missing diagram in the doc but there is beautiful one for basic schemas.

How do you do to make your png image format ?

Thanks for your help
Jmrenouard@gmail.com

bennuttall, 2015-11-01:
We have a great illustrator @samalder who makes them.

closed 2015-11-01 by bennuttall

jmrenouard, 2015-11-01:
Thank for your quick response 😊

bennuttall, 2015-11-02:
Feel free to use any of them you see in our [resources](https://www.raspberrypi.org/resources/) but unfortunately there's no @samalder-as-a-service.

waveform80, 2015-11-03:
@bennuttall any chance we can get the SVGs off Sam for the docs (I'm assuming they come from vectors originally)? Mainly useful for vector output like PDF and Epub.

bennuttall, 2015-11-09:
I'm sure we can - I'll ask him today