gpiozero/gpiozero #99: v1.0 release

bennuttall, 2015-11-03:
There's a new Raspbian release in two weeks and I'd like to get a 1.0 release of GPIO Zero in there.

There are various things we've pinned down as features or additions required for v1.0. I think any missing components can be left out for now, but more importantly the API needs to be finalised.

I think the following issues need resolving:

- [ ] #93 Temperature sensor 
- [ ] #92 MCP300X argument ordering
- [ ] #75 Change "Blink" to "Buzz" for buzzers
- [ ] #44 Deb packaging

I think we may pick up a few other minor details that need resolving before the release but I think this is roughly enough. Things like Energenie can come later, and can follow the existing API pattern.

waveform80, 2015-11-03:
Sure, sounds good. It went down a storm at the latest picademy - GPIO workshop was *much* smoother and I'm convinced it helped the attendees get more done during the project day simply because less time was burned debugging stuff (sudoless GPIO was also a *major* help in that respect)

bennuttall, 2015-11-04:
Great!

solderblob, 2015-11-06:
There is still a bouncetime (without underscore) on line 263 in input_devices.py
I use Python IDLE and have problems to stop a program that uses when_pressed and when_released - ctrl-z or ctrl-c doesn't stop it, have to use ctrl-d to kill IDLE Shell. Is there some other trick?

waveform80, 2015-11-06:
That `bouncetime` is meant to be there as that's what the underlying RPi.GPIO uses. We alias that to `bounce_time` in gpiozero for consistency with the rest of the API.

I'll have a look into the IDLE issue - if you've got reproduction code, could you put it in a new ticket so I can have a look at it?

solderblob, 2015-11-10:
Sorry about my wrong diagnostics of the error - but the problem is still there, perhaps it's more subtle than I thought. When I run the following (stripped down) program after booting the RPi 
from gpiozero import Button
button2 = Button(2)
def toggle2():
    pass
button2.when_pressed = toggle2

I get the error
Traceback (most recent call last):
  File "/home/pi/test.py", line 3, in <module>
    button2 = Button(2)
  File "/usr/local/lib/python3.4/dist-packages/gpiozero/input_devices.py", line 399, in __init__
    super(Button, self).__init__(pin, pull_up, bouncetime)
  File "/usr/local/lib/python3.4/dist-packages/gpiozero/input_devices.py", line 256, in __init__
    bouncetime=-666 if bounce_time is None else int(bounce_time * 1000)
RuntimeError: Failed to add edge detection

but if I make a new attempt to run the program, the error disappears. With my original slightly longer program I had to restart it three times before the error disappears. A reboot of the RPi is needed to make the problem appear again, a restart of Python is not enough to get the error.

bennuttall, 2015-11-12:
I'm having the same issue myself. I notice that every time I try to create a Button it raises `RuntimeError: Failed to add edge detection`. I have to run it twice to get it to work.

bennuttall, 2015-11-12:
I think all we need to do now is quickly fix #75, drop #93 and decide on #92 then #44 will can be done completing the release.

bennuttall, 2015-11-12:
The new Raspbian will be going out in the next week so we should be in it!

waveform80, 2015-11-14:
The failed to add edge detection issue is an issue in the underlying RPi.GPIO; we've got it as #50 which I should re-open as I've hit this a couple of times myself. I'm pretty sure it's a race condition in RPi.GPIO's event detection code (which manipulates a linked list in two threads without any mutex protection), but I'm going to need some time to track it down and squish it.

In other words, it's not an error in gpiozero per sé but we do need to get it fixed in the underlying library for the sake of ease of use. Given the complexity of the issue (it's likely a C-based multi-thread race condition), I doubt I'll track it down before 1.0. Even if I did, it's in a library we don't control, so I can't guarantee it'd be released alongside 1.0 anyway, but I'll certainly do what I can to fix it.

bennuttall, 2015-11-15:
Ok that's everything sorted. Let's check over everything, update the docs and @waveform80 you can create a package when you think it's ready.

closed 2015-11-16 by waveform80

bennuttall, 2015-11-16:
\o/