gpiozero/gpiozero #100: Music Box example doesn't work

lurch, 2015-11-04:
When I first saw the http://pythonhosted.org/gpiozero/recipes/#gpio-music-box example, I *suspected* that something didn't look right - AFAICT the locally-defined play function would never get called.

So I finally got round to breaking out my breadboard and installing gpiozero, and giving it a whirl. Upon running the code, it actually prints out an error!
`sudo python musicbox0.py` prints:
```
Traceback (most recent call last):
  File "musicbox0.py", line 21, in <module>
    button.when_pressed = sound.play
  File "/usr/local/lib/python2.7/dist-packages/gpiozero/devices.py", line 75, in __setattr__
    return super(GPIOBase, self).__setattr__(name, value)
  File "/usr/local/lib/python2.7/dist-packages/gpiozero/input_devices.py", line 163, in when_activated
    self._when_activated = self._wrap_callback(value)
  File "/usr/local/lib/python2.7/dist-packages/gpiozero/input_devices.py", line 210, in _wrap_callback
    'value must be a callable which accepts up to one '
gpiozero.input_devices.InputDeviceError: value must be a callable which accepts up to one mandatory parameter
```

and `sudo python3 musicbox0.py` prints:
```
Traceback (most recent call last):
  File "/usr/local/lib/python3.2/dist-packages/gpiozero/input_devices.py", line 197, in _wrap_callback
    inspect.getcallargs(fn)
  File "/usr/lib/python3.2/inspect.py", line 940, in getcallargs
    spec = getfullargspec(func)
  File "/usr/lib/python3.2/inspect.py", line 834, in getfullargspec
    raise TypeError('{!r} is not a Python function'.format(func))
TypeError: <built-in method play of Sound object at 0x76959308> is not a Python function

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.2/dist-packages/gpiozero/input_devices.py", line 203, in _wrap_callback
    inspect.getcallargs(fn, self)
  File "/usr/lib/python3.2/inspect.py", line 940, in getcallargs
    spec = getfullargspec(func)
  File "/usr/lib/python3.2/inspect.py", line 834, in getfullargspec
    raise TypeError('{!r} is not a Python function'.format(func))
TypeError: <built-in method play of Sound object at 0x76959308> is not a Python function

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "musicbox0.py", line 21, in <module>
    button.when_pressed = sound.play
  File "/usr/local/lib/python3.2/dist-packages/gpiozero/devices.py", line 75, in __setattr__
    return super(GPIOBase, self).__setattr__(name, value)
  File "/usr/local/lib/python3.2/dist-packages/gpiozero/input_devices.py", line 163, in when_activated
    self._when_activated = self._wrap_callback(value)
  File "/usr/local/lib/python3.2/dist-packages/gpiozero/input_devices.py", line 210, in _wrap_callback
    'value must be a callable which accepts up to one '
gpiozero.input_devices.InputDeviceError: value must be a callable which accepts up to one mandatory parameter
```

(I'm testing on Wheezy, hence the requirement for `sudo`).
Perhaps to a beginner, it might not be clear that the "up to one mandatory parameter" error message also includes the option of zero parameters?

It looks like the error is coming from https://github.com/RPi-Distro/python-gpiozero/blob/master/gpiozero/input_devices.py#L187 - observe this output from an interactive python prompt:
```python
>>> import pygame.mixer
>>> pygame.mixer.init()
>>> drum = pygame.mixer.Sound("samples/drum_tom_mid_hard.wav")
>>> type(drum.play)
<type 'builtin_function_or_method'>
>>> callable(drum.play)
True
>>> import inspect
>>> inspect.getcallargs(drum.play)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python2.7/inspect.py", line 900, in getcallargs
    args, varargs, varkw, defaults = getargspec(func)
  File "/usr/lib/python2.7/inspect.py", line 815, in getargspec
    raise TypeError('{!r} is not a Python function'.format(func))
TypeError: <built-in method play of Sound object at 0x76d5e120> is not a Python function
```

I **was** able to get the GPIO Music Box example working by modifying it to:
```python
from gpiozero import Button
import pygame.mixer
from pygame.mixer import Sound
from signal import pause

pygame.mixer.init()

def play(button):
    pin = button.pin
    sound = sound_pins[pin]
    print("playing note from pin %s" % pin)
    sound.play()

sound_pins = {
    2: Sound("samples/drum_tom_mid_hard.wav"),
    3: Sound("samples/drum_cymbal_open.wav"),
}

buttons = [Button(pin) for pin in sound_pins]
for button in buttons:
    button.when_pressed = play

print("ready")
pause()
```

This is only the first time I used gpiozero, but it seems counter-intuitive that the `play` callback is actually slightly more complex than the [RPi.GPIO equivalent](https://www.raspberrypi.org/learning/gpio-music-box/worksheet/), due to having to unwrap the pin number from the button object? But perhaps this allows additional flexibility.

waveform80, 2015-11-04:
Sorry, this one's my fault - I didn't realise the type introspection
doesn't work on external functions (though that shouldn't surprise me).
It's possible to do it more simply by not unwrapping the button's pin (just
wrap the play() method in a python proxy like a lambda), but that shouldn't
be necessary. I'll get this fixed up (although it's probably going to
require a bit of a kludge to support external calls like assuming they
don't take parameters).

Dave.

closed 2015-11-04 by waveform80

lurch, 2015-11-19:
So, now that this has been fixed (hurrah!) the locally-defined `play(pin)` function is indeed superfluous (as I initially expected), and should be removed to avoid confusion. Should I submit this as a new issue?

Also, this particular recipe, as well as a number of others on the Recipes page, seem to be missing the pause() at the end of the script?

waveform80, 2015-11-19:
Oh, good point! Yes please :)

lurch, 2015-11-19:
PR #120

commit e51f3e19 (2015-11-04): Merge pull request #102 from waveform80/external-callbacks

Fix #100
commit e6be9ab2 (2015-11-04): Fix #100

Permit built-in (C implemented) methods and functions to be used as
event callbacks. Given that prototype introspection is not available for
such routines, assume that they take no (mandatory) parameters.