gpiozero/gpiozero #107: Motor pin numbers not exposed as public properties

bennuttall, 2015-11-15:
Most (all?) components provide a `pin` property containing the pin number the device is connected to. However, `Motor`'s implementation hides them.

I propose we expose them as either `motor.forward_pin` and `motor.backward_pin` or `motor.pins`:

```python
>>> motor = Motor(forward=17, backward=18)
>>> motor.forward_pin
17
>>> motor.backward_pin
18
```

or

```python
>>> motor = Motor(forward=17, backward=18)
>>> motor.pins
(17, 18)
```

I think I prefer the former.

waveform80, 2015-11-15:
Hmm ... actually, I'd rather expose the device and let people grab the pin from that. Firstly that's consistent with `LEDBoard`, `TrafficLights`, etc. exposing their LED devices, but it allows people to query the motor values directly (should they wish). In other words:

```python
>>> motor = Motor(forward=17, backward=18)
>>> motor.forward_device
<gpiozero.PWMOutputDevice object on pin=17, active_high=False, is_active=False>
>>> motor.forward_device.pin
17
>>> motor.backward_device.value
0.0
```

For that matter, we should probably expose the motor devices on the `Robot` class. For example:

```python
>>> robot = Robot((17, 18), (2, 3))
>>> robot.left_motor.forward_device.pin
17
```

Hmmm, now that I think about it, `CompositeDevice` needs a proper `__repr__` implementation too.

bennuttall, 2015-11-15:
Ok sure

commit c929b9a5 (2015-11-15): Fix #107

Add forward_device and backward_device to Motor, left_motor and
right_motor to Robot, and ensure all CompositeDevice descendents have a
proper close() method and closed property. Also, add a few more
_check_open calls around the place to make sure GPIODeviceClosed is
properly raised in response to read and writing values.
commit ef9e5802 (2015-11-15): Merge pull request #108 from waveform80/motor-pins

Fix #107
closed 2015-11-15 by waveform80