gpiozero/gpiozero #109: Line Follower

bennuttall, 2015-11-15:
There seem to be two types of line followers out there commonly in use: one uses two GPIO pins (both used for input), the other uses one pin.

There are plenty of straightforward RPi.GPIO implementations. I'm not sure how effective they are.

Single GPIO pin example (from [camjam kit 3 worksheet](https://drive.google.com/file/d/0B61nxMOqZJ0YdUkwRUlBRkRUTWs/view)):

```python
follower = 25

GPIO.setup(follower, GPIO.IN)

while True:
    if GPIO.input(follower) == 0:
        print('The sensor is seeing a black surface')
    else:
        print('The sensor is seeing a white surface')
sleep(0.2)
```

Two GPIO pin example (provided by @RPi-Yaz14 [here](https://github.com/RPi-Yaz14/gpiozerowiimote/blob/master/linefollower.py)):

```python
left = 21
right = 26

GPIO.setup(left, GPIO.IN)
GPIO.setup(right, GPIO.IN)

while True:
    if GPIO.input(left) == GPIO.input(right):
        print("forward")
    if GPIO.input(left) != 0:
        print("left")
    if GPIO.input(right) !=  0:
        print("right")
```

I've a feeling the single GPIO one is doing the same thing (reading on two inputs) but returning a single value.

I did have a two-input one but I gave it to Sam to give you @waveform80. (maybe there's another one around somewhere).

Should we create one class or two, for the different types of sensor?

bennuttall, 2015-11-15:
Also maybe it should be called Line Detector, as Line Follower suggests following - but that's the robot's job!

bennuttall, 2015-11-15:
Not sure which, if any, of our base classes this should be based on.

RPi-Yaz14, 2015-11-15:
Base it on the first one (1 GPIO) as the Edukit uses the individual sensor, so people can add extras and just assign them to different pins. I used the Ryanteck 3 sensor PCB, hence used 2 pins.

bennuttall, 2015-11-15:
Is Ryan's one actually two line detectors in one?

RPi-Yaz14, 2015-11-15:
It has 3 (there is one for centre which I didn't really see as that relevant)

RPi-Yaz14, 2015-11-15:
His detectors are all individual sensors on one PCB, using a different pin for each

waveform80, 2015-11-15:
Okay, firstly it should definitely be LineDetector or LineSensor or something of that ilk rather than follower. As you note, that connotes purpose which is down to the code the user associates with the sensor. As to whether we use separate classes - I'd say probably one class per class of line sensor but there might be a hierarchy if we can re-use stuff. I should probably go and read up on this a bit more when I get a second!

bennuttall, 2016-02-10:
Just realised `LineSensor` was in the release but undocumented.

bennuttall, 2016-02-10:
We should add at least one variant, e.g. `DualLineSensor` as discussed previously (somewhere, sometime)

lurch, 2016-02-10:
Perhaps `DualLineSensor` should be a `CompositeDevice` of two `LineSensor`s?

AIUI with a single line-sensor you can only detect whether you're over the line or not, so if you 'lose' the line you have no idea whether you need to turn left or right to get 'back on' the line. But with two or more line-sensors you'll hopefully have at least one sensor still over the line, while another sensor isn't over the line, so you have a better idea of which direction to turn, to get back on the line.

waveform80, 2016-04-02:
The composite idea for multi-sensors is nice, especially as we could do another -1 to +1 value for that (0 for on the line, -1 for "off the left", +1 for "off the right" or something ... how to handle off the line entirely though ... not sure, have to think about that). For now, for the 1.2 release I'm just going to document the single LineSensor class properly. As @RPi-Yaz14 notes, that's the one in the CamJam kit, so we can presume it's the common one and it can be used as a basis for multiple sensor devices until we get around to doing something more refined with it.

waveform80, 2016-04-02:
Yup, I think you're absolutely right - after a bit of thought the compound value (of all the individual sensor values) which a `CompositeDevice` derivative would produce anyway is more useful and more representative of the sensor's state (unlike an H-bridge attached to a motor where -1 to +1 makes some physical sense)

commit 00373b60 (2016-04-02): Merge pull request #244 from waveform80/line-follower

Fix #109
commit 23a63697 (2016-04-02): Fix #109

Document and test the LineSensor class properly. Multi-sensor devices
can wait for now.
closed 2016-04-02 by waveform80