gpiozero/gpiozero #116: Use different pin for LED examples?

lurch, 2015-11-19:


waveform80, 2015-11-19:
Hmm, good point. @bennuttall any chance we can get the wiring diagram revised? Or that we could get the original SVG (so we could revise it ourselves?)

lurch, 2015-11-19:
??
The wiring diagram (on http://pythonhosted.org/gpiozero/outputs/ ) *does* use GPIO 17, which is fine.
I was suggesting changing the recipes on http://pythonhosted.org/gpiozero/recipes/ (which don't have any pictures) to use GPIO 17 to match.

waveform80, 2015-11-19:
Doh - I was looking the button example! Serves me right for not concentrating on things fully!

lurch, 2015-11-24:
Looks like somebody else agrees with me ;-)
https://www.raspberrypi.org/blog/gpio-zero-a-friendly-python-api-for-physical-computing/#comment-1248535

bennuttall, 2016-01-07:
I never replied to this - sorry. Yes, we should probably avoid the pins with nasty surprises.

@samalderhyde says the SVGs don't render correctly for some reason but he can generate all the diagrams we need. I'll give him a list when he's in a less busy period.

waveform80, 2016-01-31:
The SVGs don't render correctly? Hmm, send 'em my way anyway; that's weird enough I'd like to the see them and they may be fixable with Inkscape (I'm betting Sam's using AI which, while a very powerful package, has rather sucky SVG handling).

PR #153 (2016-01-31): Move LED example from GPIO2 to GPIO17
Fixes #116
closed 2016-01-31 by waveform80