gpiozero/gpiozero #118: Internal state not getting properly cleaned up on exit?

lurch, 2015-11-19:
I dunno if this is a problem with gpiozero, or a problem with RPi.GPIO, or even whether it's not a bug at all and is "expected" behaviour, but I noticed something weird this evening... and managed to narrow it down to a very simple example.

I have three test scripts...
led_pause.py:
```python
from gpiozero import LED
from signal import pause

led = LED(17)
pause()
```

led_on.py:
```python
from gpiozero import LED

led = LED(17)
led.on()
```

led_off.py:
```python
from gpiozero import LED

led = LED(17)
led.off()
```

When I first run `sudo python led_pause.py` after powering up the Pi, the LED stays off as expected (and remains off after I Ctrl-C). When I run `sudo python led_on.py`, the button very briefly flickers on and off again, as explained by http://pythonhosted.org/gpiozero/notes/#keep-your-program-alive-with-signalpause .
But if I now run `sudo python led_pause.py` the LED immediately lights up?! When I Ctrl-C, the LED goes off again, but running led_pause.py again turns on the LED. Only if I run `sudo python led_off.py` does the LED remain off when I next run led_pause.py

waveform80, 2015-11-19:
This is due to the way the GPIO hardware works under the covers. GPIOs (mostly, with a couple of exceptions) default to being "inputs". When they're set to inputs without any pull-up, the pin will be "low" (except for GPIOs 2 and 3 which have hardware pull-ups attached). When set to an "output" you might expect a GPIO to default to "low" but actually it'll immediately take *whatever the last output set was*. This is stored in a GPIO register and is not cleared when a GPIO is changed back to input.

So, just to be clear, what's happening is the following:

1. You boot up the Pi. GPIO17's function select register is currently "input", and the GPIO17 output value register is 0.
2. You run `led_pause.py` and GPIO17's function select register is set to "output". The pin is still low because the value register is still 0.
3. The script ends and GPIO17's function select register is set back to "input" by the GPIO.cleanup function.
4. You run `led_on.py`. GPIO17's function select register is set to "output", and the value register is set to 1.
5. The script ends and GPIO17's function select register is set back to "input" by GPIO.cleanup. Note that the value register *is still 1* but is currently ignored because the function select register is set to "input".
6. You run `led_pause.py` again, and GPIO17's function select register is set to "output". The pin is immediately set high because the value register is 1.

Now ... I could extend the GPIO cleanup to include resetting value registers as well. I'm not sure how that sort of patch would go down as I do know a few cases of people relying on this functionality (i.e. that the GPIO pins effectively "remember" their last states between runs). There's also an interesting note about pull-ups in the BCM2835 docs (which I've been trawling while digging into RPi.GPIO's guts):

> Note that it is not possible to read back the current Pull-up/down settings and so it is the
users’ responsibility to 'remember' which pull-up/downs are active. The reason for this is
that GPIO pull-ups are maintained even in power-down mode when the core is off, when
all register contents is lost.

In other words, assuming I'm reading this correctly, pull-up status even survives reboots!

Anyway, would you like to see output values reset upon shut down? If so, I might have a way of doing it without affecting RPi.GPIO (basically: do it in gpiozero in `OutputDevice.close` - but I'll need a bit of time to experiment with it as there's all sorts of horror involved when calling that from `__del__`).

lurch, 2015-11-19:
Hmmmm, interesting....

> Anyway, would you like to see output values reset upon shut down?

I guess that's one possibility, with the other option being to automatically clear the output value each time an OutputDevice is created.
I was thinking earlier that an LED turning on unexpectedly isn't too much of a problem, but if the same thing happens with a Motor object, the result could be pretty disastrous!

Whichever way you tackle this, having the last-GPIO-state being "remembered" between program runs is certainly something I'd expect gpiozero to "protect" the user from. Principle of least surprise and all that ;-)
Hmm, maybe you could have a dont_reset_value parameter to the \_\_init\_\_ of OutputDevice, for advanced users (who *do* wan't the state to be remembered) to make use of?
This is obviously a discussion for you and @bennuttall to have :)

> There's also an interesting note about pull-ups in the BCM2835 docs ... In other words, assuming I'm reading this correctly, pull-up status even survives reboots!

From what I understand that *used* to be the case, but since https://www.raspberrypi.org/documentation/configuration/pin-configuration.md got added, the firmware resets everything to a known state before loading the Linux kernel. There was a big discussion about it in the comments section on http://raspi.tv/2013/rpi-gpio-basics-3-how-to-exit-gpio-programs-cleanly-avoid-warnings-and-protect-your-pi (which include links to additional info).

bcroston, 2015-11-19:
What about using the initial= parameter of GPIO.setup() - this was added to avoid a blip on a motor like has been mentioned.

waveform80, 2015-11-19:
Hmmm, interesting information on the device tree stuff - thanks! Just skimming the comments on that article (incidentally I think there's a little mistake in the article where it claims GPIO.cleanup() "... only affects any ports you have set in the current program" - I'm pretty sure that's not the case, having read that code recently - maybe it was at some point in the past though).

Anyway, it does sound like the `initial` parameter is something we ought to expose in the base OutputDevice class. I'll have a look at that a bit later this week.

lurch, 2015-11-19:
> Anyway, it does sound like the initial parameter is something we ought to expose in the base OutputDevice class.

I was just looking at the code and thinking the same. But I guess you'll want to "massage" the `initial` value given to `OutputDevice.__init__` (with it defaulting to 'off' IMHO) before passing it to `GPIO.setup()`, in order to "play nicely" with the `active_high` parameter? And I also assume that all sub-classes of OutputDevice will need to expose this new param too?

Hmmm, looking at the rest of the code in `output_devices.py` I can't quite tell why `PWMOutputDevice` inherits from `DigitalOutputDevice` rather than `OutputDevice` ?

waveform80, 2015-11-19:
Yeah, I'll have to look at the interaction of `initial` with `active_high`. And yes, all subclasses will need to grow an optional param on their `__init__` methods too.

As to why `PWMOutputDevice` doesn't inherit from `DigitalOutputDevice` this is mostly because `DigitalOutputDevice` only adds one piece of useful functionality: `blink`. At some point my intention is to add something similar to `PWMOutputDevice` but it's going to be rather different to the one in `DigitalOutputDevice` (my hunch at the moment is that it'll be something closer to pulse in the Explorer HAT library with an on and off time, along with rise and fall time - but I'm not 100% sure about the design yet). So, given the intention is to override the one useful thing `DigitalOutputDevice` provides, there's not much point in inherting from it (semantically, I'm also not sure it should inherit from `DigtialOutputDevice`; PWM devices feel like a different branch to me - somewhere between digital and analog territory).

lurch, 2015-11-19:
You misread my message again... ;-)

Currently `PWMOutputDevice` *does* inherit from `DigitalOutputDevice` https://github.com/RPi-Distro/python-gpiozero/blob/master/gpiozero/output_devices.py#L215 but I agree with you that this doesn't make much sense, so I was asking why it doesn't just inherit from `OutputDevice`. Glad to see I wasn't barking up the wrong tree!
Let me know if you'd like me to submit this inheritance-change as a PR - I guess the `self._stop_blink()` should also then be removed from https://github.com/RPi-Distro/python-gpiozero/blob/master/gpiozero/output_devices.py#L264 as well.

lurch, 2015-11-19:
...and if I did do that, would I also need to have the `on()` and `off()` methods in `PWMOutputDevice` (to turn things fully-on and fully-off) ?

waveform80, 2015-11-19:
Dammit ... I should have coding days and answering tickets days and not try to intersperse them (context switching is *hard*)!

Given that's the way round it is, it's probably only that way to provide `blink` functionality on `PWMOutputDevice` until I can get around to providing the better replacement (the intention is to keep the first two parameters the same, `on_time` and `off_time` for compatibility with existing code, but then extend the prototype with more stuff). I wouldn't worry about fixing it at the moment, as it'll just remove functionality that someone might be relying on (now we're at 1.0 we should start taking API stability seriously).

lurch, 2015-11-19:
> I wouldn't worry about fixing it at the moment, as it'll just remove functionality that someone might be relying on (now we're at 1.0 we should start taking API stability seriously).

Fair enough. Is it worth opening up a "reminder" issue, that at some point it'd make sense to refactor `output_devices.py` to change `PWMOutputDevice` to inherit directly from `OutputDevice` (without breaking the 1.0 API)? Or is this now destined to remain as-is forever? ;)

waveform80, 2015-11-19:
Yeah, definitely worth opening a reminder issue. I don't have any issue with changing the inheritance of PWMOutputDevice; technically that's an API break but I can't imagine a use-case for checking `isinstance(d, DigitalOutputDevice)` and expecting it to come back `True` specifically for `PWMOutputDevice` descendents (whereas I can for testing the further up `OutputDevice` in order to determine whether or not something is an output or input device). In other words, provided the `blink` method can be called with the same parameters as before I'm happy to declare that "compatible enough".

lurch, 2015-12-08:
Not sure if it's relevant, but I just found another nugget of info about pullup states and warm-boots: https://github.com/raspberrypi/firmware/issues/487#issuecomment-148700352

lurch, 2016-01-31:
So... going back to the `initial` stuff - is this something that could be squeezed into the 1.1 milestone?

waveform80, 2016-01-31:
Yup, this ought to go in 1.1 - I can squeeze it in.

waveform80, 2016-01-31:
Okay, just thinking about this. I think the parameter should be called `initial_value` (makes it blindingly obvious we're talking about an initial state for the device rather than anything else) and that it should default to `False`. However, I'm thinking it should literally be the initial value for the `value` property; i.e. when active_high=True (the default), set the pin low but when active_high=False set the pin high so that the device itself remains off (when playing around with the PWM active high stuff I had a LED wired up active low and noted every time I instantiated a normal LED class on it, it lit up).

The default should play nicely with PWMOutputDevice as well since False==0.

lurch, 2016-01-31:
Yup, that's pretty much what I was thinking in https://github.com/RPi-Distro/python-gpiozero/issues/118#issuecomment-158049019

Although thinking about it some more, in an earlier comment you said "I do know a few cases of people relying on this functionality (i.e. that the GPIO pins effectively "remember" their last states between runs)" so maybe the `initial_value` parameter should also accept a value of `None`, to indicate that no `initial` value at all should be passed to RPi.GPIO ?

waveform80, 2016-01-31:
My thoughts precisely - just coding it now :)

waveform80, 2016-01-31:
Though I have just realized that `value` is completely screwed up in OutputDevice when active_high is False. I'll fix all this up while I'm at it.

commit 8b215992 (2016-01-31): Fix #118

... and fix up active_high values while we're at it ...
commit 96a91128 (2016-01-31): Merge pull request #157 from waveform80/output-initial-value

Fix #118
closed 2016-01-31 by waveform80