gpiozero/gpiozero #119: Suggestion for Recipes page

lurch, 2015-11-19:
In the http://pythonhosted.org/gpiozero/recipes/#full-colour-led-controlled-by-3-potentiometers example you use both the 'manual way' of doing things, and also a 'cleaner' example connecting the `source` of an LED to the `values` of an ADC.

Maybe you could also illustrate a similar thing in the http://pythonhosted.org/gpiozero/recipes/#button-controlled-led example by adding e.g.
```python
from gpiozero import LED, Button
from signal import pause

led = LED(2)
button = Button(3)

led.source = button.values

pause()
```

(at least, I assume that's the intention for the `source` and `values` properties?)

lurch, 2015-11-19:
Slightly off-topic:
I'm not very familiar with generators, is the code below the "correct" way to set up an "inverse generator" so that when the button is pressed, the LED is off?

```python
from gpiozero import LED, Button
from signal import pause

led = LED(2)
button = Button(3)

def invert(gen):
    for value in gen:
        yield not value

led.source = invert(button.values)

pause()
```

(of course in this particular example, the code using when_pressed / when_released would actually be a bit shorter)

waveform80, 2015-11-19:
Yup, that's the intention of source and values. I really must find some time to write up some good examples of using them, but both your examples are good ones. That `invert` filter is perfect for boolean valued components (button, LED, etc.) but is a bit "binary" for the more nuanced components like LightSensor which return a value somewhere between 0 (dark) and 1 (light). In that case you might want something like this:

```python
from gpiozero import PWMLED, LightSensor
from signal import pause

sensor = LightSensor(18)
led = PWMLED(2)

def invert(gen):
    for value in gen:
        yield 1 - value

led.source = invert(sensor.values)
```

Now the LED should be bright when the sensor is dark and dark when the sensor is lit. You can also do neat things like make a "slow" filter, and a "delay" filter:

```python
from time import sleep

def slow(gen):
    for value in gen:
        yield value
        sleep(0.1)

def delay(gen, n=10):
    queue = []
    for value in gen:
        queue.append(value)
        if len(queue) > n:
            yield queue.pop(0)
```

Now you can have a second LED and have its value set to the same value as the first LED, but a bit delayed:

```python
led2 = PWMLED(3)

led2.source = delay(slow(led.values))
```

You can also make filters that work like logic gates using `zip`:

```python
# This is required for python 2 (and does nothing in python 3 which already has an
# iterable zip built in - so you could omit this on python 3)
try:
    from itertools import izip as zip
except ImportError:
    pass

def and_gate(gen1, gen2):
    for v1, v2 in zip(gen1, gen2):
        yield v1 and v2
```

Now if you had two buttons you could set an LED to come on only if both buttons are pressed:

```python
btn1 = Button(17)
btn2 = Button(18)
led = LED(4)

led.source = and_gate(btn1, btn2)
```

I used versions of this the other day to demonstrate building a half-adder and a full-adder using some buttons (for the inputs) and a couple of LEDs (for the outputs). You can also extend the `and_gate` recipe to have as many inputs as you like:

```python
def and_gate(*gens):
    for values in zip(*gens):
        yield all(values)
```

And similarly you could use `any` for an n-input `or_gate`. Anyway, if you fancy doing some doc patches, there's some ideas! I will get around to it at some point but I'm diving into the guts of RPi.GPIO at the moment, so it might be a while before I come up for air!

lurch, 2015-11-19:
Thanks for the info and examples! Gives me lots more to experiment with.

> Now the LED should be bright when the sensor is dark and dark when the sensor is lit.

Perfect for a "night light" type application :-)

> And similarly you could use any for an n-input or_gate.

Perfect for a "burglar alarm" type application :-)

Something else I also noticed while experimenting is that https://github.com/RPi-Distro/python-gpiozero/blob/master/gpiozero/devices.py#L136 seems to be running as fast as possible, which means Python ends up consuming 100% CPU. I guess if you added a very small time.sleep() in there, then it should reduce the CPU usage without introducing any user-visible slowdown? Or if that would be too detrimental, maybe you could provide some method of having the user tell the 'source' property how long it should sleep before fetching the next generator value?

lurch, 2015-11-19:
...and for another idea, I guess you could plug the output of a random-value-generator into the source of a PWMLED, to give a "flickering candle" effect? :)

waveform80, 2015-11-19:
Yup, what actually happens when you assign something to "source"? Behind the scenes, some fairly complex stuff actually goes on:

1. If there's an existing source, we shut down the thread reading it and wait for it to terminate
2. If the new source is `None`, do nothing, exit here
3. If the new source isn't `None`, start a new background thread that reads values from the assigned iterable as fast as possible until it's exhausted

If you throw the `slow` filter above around the new source you'll find it won't eat 100% of the CPU time. So why don't we do that by default? Firstly because it doesn't actually matter that it's eating 100% of CPU time (it's not really keeping the CPU that busy - it's more that CPU time is measure of the number of threads "doing work" - the Pi would be perfectly responsive even if you chained several of those together). Secondly because it's easy to throw in a `slow` filter if it really bothers anyone, and thirdly because leaving it unfiltered by default gives users the most flexibility.

For example, at the moment one can make nice effects like smoothly fading in and out LEDs using source. But if we introduce a default delay we potentially remove that capability (depending on its length). Could we make the default delay variable in length? Yes, but only by adding another property to the class (and personally I think the filter method is much cleaner and more flexible).

Here's some more interesting filter ideas: one could write a filter which only returns every other value from the source:

```python
def every_other(gen):
    state = True
    for value in gen:
        if state:
              yield value
        state = not state
```

Or a more subtle one that only yields values on a signal from another component, like a clock sync:

```python
def clocked(gen, clk):
    gen = iter(gen)
    for value in clk:
        if value:
            yield next(gen)
```

Now you can demonstrate using a clock signal to "copy" values from one LED to another:

```python
led1 = LED(12)
led2 = LED(16)
btn = Button(26)

led2.source = clocked(led1.values, btn.values)
```

Set led1 on, and led2 won't copy that state until you press the button. Likewise, if you then set led1 off, led2 won't copy that state until you press the button again (the button is the "clock" signal).

waveform80, 2015-11-19:
Oooh - nice idea on the random values one - I hadn't thought of that!

```python
from random import random

def random_source():
    while True:
        yield random()
```

waveform80, 2015-11-19:
And then:

```python
led = PWMLED(12)
led.source = random_source()
```

Or if you want a binary version:

```python
def random_source():
    while True:
        yield random() > 0.5

led = LED(12)
led.source = random_source()
```

lurch, 2015-11-19:
Okay... I think I'm starting to get the idea :)  But not quite up to writing documentation about it yet ;-)

I guess the idea with the 'slow' filter, is that it's easier to slow down the rate at which the generator emits values, than it is to slow down the rate at which the source requests values? And I guess it'd also be easy to add a 'delaytime' parameter to the 'slow' filter to make it more adjustable, i.e.
```python
def slow(delaytime, gen):
    for value in gen:
        yield value
        sleep(delaytime)
```

Which, if I've understood things correctly, means you could potentially write the "blink" example as:
```python
from gpiozero import LED
from signal import pause

def slow(delaytime, gen):
    for value in gen:
        yield value
        sleep(delaytime)

def alternate():
    value = False
    while True:
        yield value
        value = not value

led = LED(17)
led.source = slow(1, alternate())
pause()
```

Lots of experiments with LED-flashing to be done this evening... :+1:

waveform80, 2015-11-19:
You certainly could write blink that way (as soon as I'd added source and values I started seeing ways to reimplement everything we were doing with events with source and values instead, but we've left events in because they can be clearer sometimes).

Anyway, the one misconception in your comment above is that there is a "rate" at which values produces items. Rather, there is a rate at which source attempts to read values which, by default, is "as fast as possible". However, the magic of generators in python is that the act of source attempting to read something from the provided iterator causes values to read the object's state.

In other words, when values is not connected to anything, it reads nothing. When you ask values for an item, it will produce one in response but won't do any further work unless you ask for another (and so on). The "slow" function works effectively by causing values to take a long time to return an item (source is trying to read as fast as ever but each time it reads a value it takes several milliseconds to get one, but we don't need to worry about queueing anything because values won't read anything more until source asks for the next value).

Hopefully that makes some sense but if not, let me know and I'll try and clarify (probably badly!)

lurch, 2015-11-19:
> (as soon as I'd added source and values I started seeing ways to reimplement everything we were doing with events with source and values instead, but we've left events in because they can be clearer sometimes).

Yeah, I guess like a lot of simple vs. advanced language features, events are simpler and easier to grok, but generators are more complex but much more powerful ;-)

> Hopefully that makes some sense but if not, let me know and I'll try and clarify (probably badly!)

http://stackoverflow.com/questions/231767/what-does-the-yield-keyword-do-in-python provides a few useful tidbits here & there.

Going back to the "slow" filter / function / generator, apart from the very first time that it's called (yields), does it make any difference if it's written as 
```python
def slow(delaytime, gen):
    for value in gen:
        yield value
        sleep(delaytime)
```
or
```python
def slow(delaytime, gen):
    for value in gen:
        sleep(delaytime)
        yield value
```
?

Presumably the "magic of generators" means that the source reading the generator gets the 'current value' of the generator (e.g. whether the button is pressed or not) at the moment of the source thread getting the value (if you see what I mean - kinda hard to explain!) regardless of how big our delaytime is?

waveform80, 2015-11-19:
> Going back to the "slow" filter / function / generator, apart from the very first time that it's called (yields), does it make any difference if it's written as [sleep before or after]?

You're absolutely right: other than the first time it's called, it makes absolutely no difference which way it's written.

> Presumably the "magic of generators" means that the source reading the generator gets the 'current value' of the generator (e.g. whether the button is pressed or not) at the moment of the source [...]

Sorry, no that's not quite right. To answer this I'm afraid I'll have to delve into how generators actually work. The trivial method of explaining them is that they're a special case of co-routines where the routine can only be entered at the top, or immediately after the last yield point during iteration. Unfortunately, if you don't know what co-routines are that's probably meaningless, so I'll try an expanded version below...

A normal function has a trivial means of execution:

```python
def foo(a, b):
    result = a + b
    return result

print(foo(1, 2))
```

1. `foo` is called with parameters 1 and 2.
2. The parameter's names are locally bound to the parameter's values (in the function's local state, `a=1` and `b=2`).
3. The function's instructions are executed from top to bottom (we calculated `result`, then return it)
4. Once the function returns, its local state is forgotten.

A generator function is rather different:

```python
def foo(a):
    yield a
    a += 1
    yield a
    a += 1
    yield a

for value in foo(1):
    print(value)
```

1. Firstly note that a generator function must be called in the context of an iterable (e.g. a `for` loop as above, in a list comprehension, or as a parameter of a function that will loop over it, like `sum()`)
2. The generator function's call starts off like the traditional function above. Parameter values are bound to their names in the function's local state (here `a=1`)
3. We start executing the function's body from top to bottom. The first statement is `yield a`. At this point, the function "yields" the value 1 to the calling loop.
4. The loop's body now executes with the returned value (1) assigned to the loop variable (`value`).
5. Now we're back at the top of the loop, we *re-enter* the generator function at the point we left it and continue executing.
6. Note that its local state (`a=1`) has been preserved. We now add 1 to `a` (local state is now `a=2`) and once again yield a value.
7. The loop's body now executes again with the yielded value (2) assigned to the loop variable.
8. We're back at the top of the loop and we re-enter the generator function again.
9. Once again, the function's local state has been preserved. We add 1 to `a` and yield a final time.
10. The loop's body now executes again with 3 assigned to the loop variable.
11. We're back to the top of the loop and we re-enter the generator function.
12. The generator function is now at the end of the function, which implicitly raises a `StopIteration` exception. The calling loop catches this and silently terminates.

The points to take away are:

* The generator doesn't have a "current value". Whenever a loop calls it, it executes code which eventually either yields another value, or raises `StopIteration` to terminate the calling loop.
* The generator's state is preserved between yields. You can think of it as a slightly odd function which starts at the top, but then "pauses" when it yields and carries on after the yield when the calling loop next iterates.
* But note that calling the generator from a different loop creates a new "instance" of that generator which will start at the top (a generator can be in use from multiple loops simultaneously and each loop represents a different runtime state for the generator).
* What happens if the calling loop terminates (e.g. `break`) whilst the generator function has yielded control to it? In that case the generator function's state is terminated too (there's some complexity here if the generator function was in the middle of a `try..finally` block in which case the `finally` clause will execute prior to the calling loop's termination, but that's about it).

You can probably guess from all this that the implementation of `values` in gpiozero is ridiculously trivial:

```python
@property
def values(self):
    while True:
        try:
            yield self.value
        except GPIODeviceClosed:
            break
```

Although it's just occurred to me I should have written it even more simply:
[... 12 lines cut ...]
```python
# ...
for v in source:
    self.value = v
    if self._source_thread.stopping.wait(0):
        break
# ...
```

So looking at a trivial thing like `led.source = btn.values` we'd see the following sequence of code:

1. In source: `for v in source` starts the values generator
2. In the values generator we start an infinite `while True` loop
3. The values generator yields the button's state (let's call it 0)
4. 0 is assigned to the source for loop's variable `v=0`
5. The source loop's body is executed and the value of `v` is assigned to the target object's value (the led is set off)
6. We haven't requested the source loop terminate so now we're back at the top of the for loop
7. We re-enter the generator and we're back in the infinite while loop which loops around and yields another value (let's call it 1 this time)
8. 1 is assigned to the source for loop's variable `v=1`
[... cut ...]
12. Because the calling loop has terminated, the `values` generator function (which was in the middle of an infinite `while True` loop) is also terminated. There's no `try..finally` statement, so nothing to clear up, and we're all done.

Okay, I've rambled on for far too long - do let me know if anything's not clear! And try working through the sequence of operations when a filter or two is thrown into the mix. The surprising thing is that it's rather easy to write powerful filters and yet when you actually dig into the order of execution, it's horribly complicated with local state lying around all over the place!

lurch, 2015-11-20:
Yeah, that's kinda how I understood things working, except I didn't realise you could have multiple yield statements in a single generator function. But now that I better understand how generators work (thanks!) even that makes perfect sense now.
Who would have thought that a library for "simpler GPIO access" would end up teaching me so much about advanced Python features?! :-)

What I meant with my previous (confusing) question, was that given an extreme example of:
```python
from gpiozero import LED, Button
from signal import pause
from time import sleep

def slow(delaytime, gen):
    for value in gen:
        sleep(delaytime)
        yield value

led = LED(17)
button = Button(3)
led.source = slow(10, button.values)
pause()
```

then when the LED gets "set" does it get set to the state of the button "now", or does it get set to the state of the button 10 seconds ago? Or (in some circumstances) does it *actually* make a difference if the `sleep` is before or after the `yield`? (contrary to your previous answer)

waveform80, 2015-11-20:
Ahh, I see. From that perspective, yes it can make a difference. Here's a quick walk-through of what will go on in your example above (with the `sleep` first in the loop):

1. Setting `led.source` spawns a background thread that runs the loop we saw previously (`for v in source: self.value = v`)
2. In this context, `source` is a call to your `slow` generator, so we start your generator
3. Now within `slow` we come to `for value in gen`. In this context, `gen` is the `btn.values` property which is itself a generator.
4. In `btn.values` we effective have `while True: yield self.value` so at this point, the button's state is read (let's say it's not currently pushed, so `False`) and yielded to the `slow` generator
5. Now we're into the `slow` generator's for-loop body. `value` is `False`, but the first thing we run across is `sleep(10)` so we delay for ten-seconds.
6. For the sake of argument, let's say that some time during that 10 second delay you press (and hold) the button.
7. Now we yield `value`. Remember that it's still `False` because we read it before the delay.
8. Finally we're back to the `source` loop. `False` is assigned to `v` and subsequently assigned to the LED's `value` property, leaving the LED off.
9. Now we're back to the top of the `source` loop so we re-enter the `slow` generator function.
10. The `slow` generator loops back round to the top of its for-loop and we re-enter the `values` generator.
11. This now yields `True` because the button is pushed.
12. Now we're back to the `slow` generator and the `value` loop variable has been assigned `True`, but we wait 10 seconds first.
13. Then we yield `True` back to the `source` loop.
14. And finally we're back in the `source` loop and the LED turns on.

So, in your example there would be a significant delay between you pushing the button and the LED turning on (assuming you pushed the button in the middle of one of the sleeps which is pretty likely).

lurch, 2015-11-20:
Thanks, that makes perfect sense, and is exactly the answer I was hoping you'd give!
I think I understand it all now :-)

So, last question for now... could my example above be re-written as:
```python
from gpiozero import LED, Button
from signal import pause
from time import sleep

def slow(delaytime, gen):
    while True:
        sleep(delaytime)
        yield next(gen)

led = LED(17)
button = Button(3)
led.source = slow(10, button.values)
pause()
```
...and then it doesn't matter where in the loop the `sleep` happens?

And I guess if I wanted to stick to using `for` rather than `while` (in case the generator is non-infinite) I could 'emulate' the sleep-before-yield version of slow, without introducing a "button lag" by rewriting it as:
```python
def slow(delaytime, gen):
    sleep(delaytime)
    for value in gen:
        yield value
        sleep(delaytime)
```

I also discovered that you can call `print` from inside generators (I wasn't sure if that would be allowed) which makes debugging things / working out what's going on a bit easier.

waveform80, 2015-11-20:
Sure, your `while` version would indeed work, as would your adjusted `for`-version.

One minor note: in certain circumstances you may need to call `iter()` prior to calling `next()`. You won't with the result of a generator like `values` but if you passed in a straight-forward list or something similar, you would. This is because `next()` expects an iterator (i.e. something that provides the special `__next__` method), while a list is a collection (which can be iterated over by deriving an iterator from it with `iter()`). So to work with all iterable inputs you'd need something like:

```python
def slow(delaytime, gen):
    gen = iter(gen)
    while True:
        sleep(delaytime)
        yield next(gen)
```

lurch, 2015-11-20:
Great, thanks.
According to https://wiki.python.org/moin/Iterator I think you meant `next` rather than `__next__` ?
(that page needs to be read *very* carefully, to distinguish iterable (e.g. a list) from iterator (which I guess you can imagine as sort of like a pointer, which automatically jumps to the next element in the list each time you read its value).)

Presumably if `gen` is already an iterator (which a generator automatically is), then `gen = iter(gen)` is effectively a nop?

waveform80, 2015-11-20:
`next` was the name under Python 2; Python 3 changed it to `__next__` as it's effectively an internal method which isn't usually called directly (only via the `next()` built-in function). And yeah, it's a hairy area linguistically! Your analogy of a pointer is an excellent one, incidentally. Just consider that each thing returned by `iter()` is a separate pointer which you then move along with `next()` and you're there.

I'm not 100% sure if `gen = iter(gen)` is a no-op in the case that `gen` is originally an iterator, or whether it winds up wrapping the iterator in another iterator. I could go and dig it out of the source code, but for now I'm happy enough to know that "it just works" (tm).

Incidentally, a while back I ran [a workshop](http://www.waveform.org.uk/presentations/generators/) for our local Python North West group on infinite iterators (and iterator filters), as I think they're hugely powerful and yet generally unknown.

Anyway, just to demonstrate a truly silly idea, here's an infinite iterator (using the "requests" and "beautiful soup" libraries; `sudo apt-get install python3-requests python3-bs4`) which polls the BBC News homepage and yields the count of the word "pie" or "pies" which I then feed to an LED to provide a visual indication of the amount of pies in the news:

```python
import re
import requests
from time import sleep
from gpiozero import PWMLED
from bs4 import BeautifulSoup
from signal import pause

def word_count(word, url="http://www.bbc.co.uk/news"):
    while True:
        page = requests.get(url)
        soup = BeautifulSoup(page.content)
        yield len(soup.find_all(text=word))
        sleep(3600) # sleep an hour before hitting the beeb again

def normalize(it):
    m = 1 # got to start somewhere north of 0
    for i in it:
        m = max(m, i)
        yield i / m

# The more pies in the news the more the LED glows!
led = LED(12)
led.source = normalize(word_count(re.compile('pies?', re.IGNORECASE)))
pause()
```

P.S. Don't ask me how long it took my warped sense of humour to come up with something politically neutral and not massively offensive like "pies" ... and don't ask what I came up with first either ...

lurch, 2015-11-20:
> next was the name under Python 2; Python 3 changed it to __next__ as it's effectively an internal method which isn't usually called directly (only via the next() built-in function).

Ahhh, that makes sense! I was wondering why `next(i)` and `i.next()` worked equally well.

Today seems to be pie-free :-/

waveform80, 2015-11-20:
Doh, that was meant to be a PWMLED ... just a sec ...

bennuttall, 2015-12-04:
@lurch has your issue been resolved or do you still have a suggestion for change?

lurch, 2015-12-05:
Sorry this got so long-winded and slightly off-topic, but I believe my [initial suggestion](https://github.com/RPi-Distro/python-gpiozero/issues/119#issue-117728022) still holds?
I guess this is a "documentation suggestion" reminder issue, but feel free to close if it's not appropriate.

waveform80, 2015-12-05:
Blame me for getting off-topic - not an un-common occurrence, sorry!

lurch, 2016-02-04:
If / when you get round to adding more examples to Recipes, I guess this is a nice example too?
http://raspi.tv/2016/using-mcp3008-to-measure-temperature-with-gpio-zero-and-raspio-pro-hat#comments

pcopa, 2016-02-04:
I second the temperature proposal. It's something I'm working on right now. How to convert a resistance/voltage to temperature.

commit 183b8547 (2016-02-07): Fix #119

Add a bunch of code snippets demonstrating generators to the recipes
chapter, including a temperature recipe for the MCP3008.
commit c5c80afe (2016-02-07): Merge pull request #170 from waveform80/gen-recipes

Fix #119
closed 2016-02-07 by waveform80