gpiozero/gpiozero #125: Adds a *args to LEDBoard on() and off() so individual pins can be controlled

martinohanlon, 2015-11-29:
At the moment, the LEDBoard class only allows you to turn on or off all the LEDs. The methods could be enhanced so optional led positions can be passed to the on and off methods to control individual leds.

martinohanlon, 2015-11-29:
Happy to assign this to myself.

waveform80, 2015-11-29:
Sounds good to me - this would allow control of multiple LEDs through a single call rather than having to make multiple calls to, e.g. `board.leds[0].on(); board.leds[1].on()` etc.

One question: should the `*args` be pin numbers as in the constructor, or LED indexes as in the `leds` property. I'd lean toward the latter for consistency with existing code (in particular I'm thinking about the classes that derive from LEDBoard) but I'd be interested in counter arguments?

Also, @bennuttall any thoughts?

martinohanlon, 2015-11-29:
Definitely indexes of the leds property. It would seem counter intuitive to access items within an abstract class by a specific value (which could change across implementations).

bennuttall, 2015-12-09:
Sorry I missed this one. I actually came here to make a similar comment myself!

I started using `LEDBoard` to control a row of LEDs and realised I'd be better off using LED in a list comprehension. The only thing `LEDBoard` provides is the ability to control them all together with on/off/blink/toggle.

I'm currently using:

```python
n = 2
leds = [LED(i) for i in (2, 3, 4)]
[led.on() for i, led in leds if n > i]
```

to turn on the first two.

waveform80, 2016-04-06:
Do we have a PR on this yet? I see @lurch has some commits which might form one, and @martinohanlon's volunteered to take this on?

Incidentally, just to counter the idea that using `value` isn't friendly - here's a little something for alternate patterns:

```python
from gpiozero import LEDBoard
from time import sleep

board = LEDBoard(5, 6, 13, 19, 26)
while True:
    board.value = (1, 0, 1, 0, 1)
    sleep(1)
    board.value = (0, 1, 0, 1, 0)
    sleep(1)
```

Or in source/values form (using the new source_delay stuff in 1.2):

```python
from gpiozero import LEDBoard
from signal import pause

board = LEDBoard(5, 6, 13, 19, 26)

def alternating():
    while True:
        yield (1, 0, 1, 0, 1)
        yield (0, 1, 0, 1, 0)

board.source_delay = 1
board.source = alternating()
pause()
```

Of course, you can make it obscure if you really want:

```python
from gpiozero import LEDBoard
from itertools import cycle
from signal import pause

board = LEDBoard(5, 6, 13, 19, 26)

def alternating():
    for i in cycle((0, 1)):
        yield tuple((j + i) % 2 for j in range(len(board)))

board.source_delay = 1
board.source = alternating()
pause()
```

Speaking of obscure, here's a knight-rider demo I threw together for fun:

```python
from gpiozero import LEDBoard
from itertools import cycle, chain
from signal import pause

board = LEDBoard(5, 6, 13, 19, 26, pwm=True)

def kitts_bumper():
    max_value = (len(board) - 1) * 10
    values = range(max_value)
    for value in cycle(chain(values, reversed(values))):
        yield tuple(
            (1 - (abs((index * 10) - value) / max_value)) ** 3
            for index, led in enumerate(board))

board.source_delay = 0.02
board.source = kitts_bumper()
pause()
```

How about counting in binary:

```python
from gpiozero import LEDBoard
from itertools import cycle
from signal import pause

board = LEDBoard(5, 6, 13, 19, 26, pwm=True)

def counter():
    for value in cycle(range(2 ** len(board))):
        yield tuple(
            bool(value & (1 << index))
            for index in range(len(board)))

board.source_delay = 0.1
board.source = counter()
pause()
```

Anyway, must sleep ... :)

lurch, 2016-04-06:
> Do we have a PR on this yet?

Um... #183

Afraid I haven't had time to rebase, but could do so tomorrow if 1.2 is imminent?

bennuttall, 2016-04-07:
If you get chance, that would be great!

waveform80, 2016-04-07:
Doh - how did I miss that? Must get more sleep...

lurch, 2016-04-07:
```python
import time
class waveform80:
    def __init__(self):
        self.sleeptime = 6
    def sleep(self):
        print("Sleeping for %f hours" % self.sleeptime)
        time.sleep(self.sleeptime * 3600)
    def kick(self):
        self.sleeptime += 1
```
:-)

#183 has now been rebased.

waveform80, 2016-04-08:
Ahh, 6 hours ... if only (tough to achieve that unbroken with a one-year old in the next room!)

waveform80, 2016-04-10:
Closed in #183

closed 2016-04-10 by waveform80