gpiozero/gpiozero #128: gpio access requires sudo

RogueM, 2015-12-02:
I have been investigating an issue I have been having with gpio zero and discussed with @bennuttall on Twitter, and after much efforts have tracked down the problem to RPi.GPIO

... I do not know what the official repo for RPi.GPIO is but I thought you might want to know about this issue and how to reproduce it, as I suspect there is a tie between those projects, and at the very least you know who to contact to address those concerns.

0) flash fresh Raspbian Jessie Lite (python3-rpi.gpio is not included)
1) install python-pip3 using apt-get (not included in Jessie Lite either)
2) install RPI.GPIO using pip3 (this will roll version 0.5.11)
3) install python3-gpiozero using apt-get

step 3 will pull python3-rpi.gpio 0.6.0a3 from the apt repo as a dependency, but 'pip list' will keep flagging version 0.5.11 as being active, even after complete reboot of the machine (not that I thought that would be required).

from there on python will not be able to access the gpio without elevated privileges, which not only affects gpiozero but also any other means to access the GPIO pins.

interestingly it only affects python3 because python-rpi.gpio 0.6.0a3 is part of the Jessie Lite image and downgrading it via pip is not possible (as least not forcefully). I haven't checked whether this problem affects the full Jessie OS image, but suspect it might.

in practical terms there is a way to resolve the problem, fairly straightforward, I guess, once you know the cause... it took me 3 days to figure out, I'm not a wiz, sure, but I doubt the average guy would even have the patience and will just use sudo - that's what I did for 3 days, hey, the spirit is weak especially when the flesh is willing! Anyhow:

4) pip3 uninstall RPi.GPIO

... this will automagically allow the RPi.GPIO 0.6.0a3 installed via apt to kick in and thereafter access to GPIO without using sudo python3 will be restored (or rather enabled in this sequence of events).

I guess that ideally the RPi.GPIO version in the pip repo should be upgraded, but I do not know where to make the request, and perhaps whoever is responsible may have a good reason to lag behind?

Anyhow, I hope this helps, let me know if there are other details you need from me, but I think reproducing the issue should be easy, as long as you follow the recipe herein.

waveform80, 2015-12-02:
Yup, I'm afraid RPi.GPIO 0.6.0a3 has been released as a deb package to the Raspbian repos, but the version on PyPI is still stuck at 0.5.11 (and PyPI installs take precedence over apt installs as far as Python is concerned). Ben Croston (@bcroston) is the maintainer of RPi.GPIO but the [project itself](https://sourceforge.net/projects/raspberry-gpio-python/) is on Sourceforge so you're probably best off opening a ticket on there.

That's the "proper" way of ultimately fixing this. However, in the meantime, if you need to work with gpiozero in a virtualenv (as I frequently do while hacking on it), and you want to avoid having to deal with sudo every time you go to test something you can try the following (I'm assuming you've got git, mercurial, build-essential, python3-dev, python3-virtualenv and, virtualenvwrapper installed and configured where appropriate):

1. Clone the gpiozero repo: `git clone https://github.com/RPi-Distro/python-gpiozero`
2. Clone the RPi.GPIO repo: `hg clone http://hg.code.sf.net/p/raspberry-gpio-python/code rpigpio` (this will get you the latest sudoless implementation)
3. Create a venv to hack on gpiozero: `mkvirtualenv gpiozero`
4. Activate the venv (if it isn't already): `workon gpiozero`
5. Install RPi.GPIO in the venv: `cd rpigpio; python setup.py install; cd ..`
6. Install gpiozero as a development package in the venv: `cd gpiozero; python setup.py develop`

Now you can hack on your checkout of gpiozero without having to deal with sudo all the damned time!

waveform80, 2015-12-02:
Sorry, I should add: ultimately this isn't a problem in gpiozero but RPi.GPIO (a separate project) which has different "latest" releases in different repos. Still, I figured it might be worth having the instructions above documented somewhere for anyone that wants to hack on gpiozero without too much pain!

RogueM, 2015-12-02:
cheers, great info, and yes, I agree this isn't a gpiozero problem. In fact after checking the full Raspbian Jessie image released recently (as opposed to the Lite version) there is next to null risk of it cropping up on a new clean SD as python3-rpi.gpio is included right off the bat.

... as soon as I can remember my sourceforge access I'll reach out for Ben Crofton to see if he has any intention to update the pip version to the latest sudoless version.

bcroston, 2015-12-02:
You can install the bleeding edge version of RPi.GPIO using pip by:

$ pip install hg+http://hg.code.sf.net/p/raspberry-gpio-python/code#egg=RPi.GPIO

I would recommend doing this within a virtualenv if possible.

alaudet, 2015-12-03:
@bcroston ...  When will the latest version of RPi.GPIO (0.6.3a right now I think) be finalized.  I did not see anything on your sourceforge page.  Any plans to move to github?

closed 2015-12-04 by bennuttall