gpiozero/gpiozero #130: Add SnowPi

bennuttall, 2015-12-03:
I'm currently working on adding in Ryan's [SnowPi](https://ryanteck.uk/raspberry-pi/114-snowpi-the-gpio-snowman-for-raspberry-pi-0635648608303.html) add-on board.

SnowPi is simply a collection of 9 LEDs, each of which can be named according to its position (nose, left, eye, right, eye, etc.)

A simple implementation could be:

```python
class SnowPi(SourceMixin, CompositeDevice):
    """
    Ryanteck SnowPi add-on board.
    """
    def __init__(self, pwm=False):
        self.nose = LED(25)

        self.left_eye = LED(23)
        self.right_eye = LED(24)

        self.left_arm_top = LED(17)
        self.left_arm_middle = LED(18)
        self.left_arm_bottom = LED(22)

        self.right_arm_top = LED(7)
        self.right_arm_middle = LED(8)
        self.right_arm_bottom = LED(9)
```

or even:

```python
class SnowPi(LEDBoard):
    """
    Ryanteck SnowPi add-on board.
    """
    def __init__(self, pwm=False):
        super(SnowPi, self).__init__(7, 8, 9, 17, 18, 22, 23, 24, 25, pwm=pwm)
```

However, I wonder if it would be possible to create groups of LEDs so that they could be accessed like so:

```python
snow_pi.eyes # both eyes
snow_pi.eyes.left # left eye
snow_pi.arms # both arms
snow_pi.arms.right # right arm
snow_pi.arms.right.top # top LED of right arm
```

Where these are akin to LEDBoard or TrafficLights, i.e. `snow_pi.arms.on()` turns all LEDs in both arms on, `snow_pi.arms.left.on()` turns the left arm on, and `snow_pi.arms.left.middle.on()` turns just one LED on.

Would that be possible? Or is it overkill?

waveform80, 2015-12-03:
You could do something like this:

```python
class SnowPiEyes(LEDBoard):
    def __init__(self, pwm=False):
        super(SnowPiEyes, self).__init__(23, 24, pwm=pwm)

    @property
    def left(self):
        return self.leds[0]

    @property
    def right(self):
        return self.leds[1]

class SnowPiArm(LEDBoard):
    def __init__(self, top, middle, bottom, pwm=False):
        super(SnowPiArms, self).__init__(top, middle, bottom, pwm=pwm)

    ...

class SnowPiArms(SourceMixin, CompositeDevice):
    def __init__(self):
        self._left = SnowPiArm(17, 18, 22)
        self._right = SnowPiArm(7, 8, 9)

    @property
    def left(self):
        return self._left

    ...

...

class SnowPi(SourceMixin, CompositeDevice):
    def __init__(self):
        self._eyes = SnowPiEyes()
        self._arms = SnowPiArms()

    @property
    def eyes(self):
        return self._eyes

    @property
    def arms(self):
        return self._arms

    ...
```

Whether it's overkill ... possibly, but I must admit the result is rather pretty code and it does provide the ability to individually address all sub-components like arms, legs, etc. if you bother to implement `value` for all the sub-classes (which would involve some new namedtuple descendents I haven't included above, etc). I could imagine several use cases for that sort of thing (snow-man's nose blinks in response to motion sensor, left and right arms light up indicating direction of ... erm ... an advent window?). Yeah, why not!

bennuttall, 2015-12-03:
Ah, I did consider the Arms/Eyes class model but wasn't sure about the namedtuple stuff. I'll have a go and see what works.

commit d1669b80 (2016-04-01): Merge pull request #240 from waveform80/snowpi

Fix #130, fix #131, fix #189
commit d6af0293 (2016-04-01): Fix #130, fix #131, fix #189

This PR implements SnowPi, adds the ability for LEDBoard's to own other
LEDBoards as well as LEDs, and enhances blink so that manually
controlling a LED automatically stops it from blinking (no matter
whether it's blinking itself or a LEDBoard is blinking it).

It also fixes up RGBLED and Motor which I managed to break with the last
PR ...
closed 2016-04-01 by waveform80