gpiozero/gpiozero #132: Can we just install this instead of upgrading to Jessie?

hippyjim, 2015-12-05:
Hi

Is there no way to just install this on Wheezy? All the "simple" instructions don't work as the package is not available. Is there a way to download it from here and install it?

closed 2015-12-05 by hippyjim

waveform80, 2015-12-05:
You can download the .debs from the [Jessie repo](http://archive.raspberrypi.org/debian/pool/main/g/gpiozero/) and `sudo dpkg -i pkgname.deb` to install it (I deliberately built the package on wheezy so it'll be compatible), or you can use `pip` with the PyPI version (`sudo pip install gpiozero`).

waveform80, 2016-01-09:
I've just checked, and it looks like the package and all the deps (including spidev) are now in the wheezy repos too so the simple install should now work there too.