gpiozero/gpiozero #135: RGBLED doesn't have blink() while docs say it does

zeenix, 2015-12-29:
I was trying out your awesome library and was slightly disapointed that RGBLED class doesn't actually have a blink() method while the docs here say, that it does:

http://pythonhosted.org/gpiozero/outputs/

It's probably cause RGBLED doesn't derive from DigitalOutputDevice like other two LED classes.

lgcproductions, 2015-12-29:
I gave up on trying to get the RGBLED working. reverted back to just using LED. :(

zeenix, 2015-12-31:
but it works fine, other than that it doesn't have a blink method. Looking at the code, it shouldn't be that hard to implement locally.

bennuttall, 2016-01-07:
I think we may have removed blink and toggle due to the nature of the device not being a simple on/off. It probably should have these methods though, it just needs to remember the state during blinking. Do regular PWM devices do this?

bennuttall, 2016-01-07:
Related to #121 I think?

waveform80, 2016-01-07:
Grrr ... again, my reST docs are correct ... but I digress (sorry, couldn't resist!). If I recall correctly, the reason that blink was removed is because it was non-trivial in the non-threaded counted case (i.e. the former implementation was simply wrong in that case ... or was it the threaded case ... anyway, one of the cases was just wrong and it wasn't an entirely simple fix so it got chopped pending further work).

zeenix, 2016-01-11:
If you are looking for a usecase, here is my pet project that needed a blink function:

https://github.com/zeenix/rasp-led-weather

You can see it in action here: https://plus.google.com/u/0/+ZeeshanAli/posts/EH6ZX1ZHn9k

You can see it in action here:

lurch, 2016-01-31:
Given that PWMOutputDevice now has a blink() method with optional fade-in/fade-out args, I wonder if RGBLED could have a blink() method with extra optional `on_colour` and `off_colour` args, defaulting to 'white' and 'black'? Just a suggestion ;-)

waveform80, 2016-01-31:
@lurch Oooh ... nice idea! Yup, that can go in 1.1.

waveform80, 2016-01-31:
Incidentally, the docs side of this should have been fixed by the rest-docs implementation (the docs now can't contain anything which the code doesn't actually implement since they're derived from the code's docstrings)

lurch, 2016-01-31:
I guess if you wanted to get super-fancy, you could even add the fade-in/fade-out args and have the RGBLED smoothly fade between the two colours! ;-)
(although it'd obviously be using RGB-fading, and so might look a bit weird when the user expects HSV-fading... hmmm.)

waveform80, 2016-02-08:
Yeah, it's RGB linear interpolation for the fading; HSV can be another ticket ;)

commit 9871ac3d (2016-02-08): Fix #135
commit a498d29b (2016-02-08): Merge pull request #174 from waveform80/rgbled-blink

Fix #135
closed 2016-02-08 by waveform80