gpiozero/gpiozero #136: button.when_pressed = whoops()

bennuttall, 2016-01-05:
Is there a way we can throw an error if someone tries to set `when_pressed` (or `when_active`) to a non-callable object? It's common to accidentally write:

```python
def hello():
    print("Hello")

button.when_pressed = hello()
```

https://twitter.com/CotswoldJam/status/684420747939438592

I'm guessing what happens here is it sets the `.when_pressed` property to `None` (the return value of the function). I imagine it would be easy enough to detect whether this value is callable and reject it if not.

@waveform80?

alexwlchan, 2016-01-07:
It looks as if passing a non-callable to `when_pressed` is already being handled. It goes through `when_activated`, which in turns calls `_wrap_callback`, and that checks if the input is callable:

```python
def _wrap_callback(self, fn):
    if fn is None:
        return None
    elif not callable(fn):
        raise InputDeviceError('value must be None or a callable')
    ...
```

[L180 of `input_devices.py`](https://github.com/RPi-Distro/python-gpiozero/blob/master/gpiozero/input_devices.py#L180)

bennuttall, 2016-01-07:
That's true. Perhaps it's not possible then?

waveform80, 2016-01-07:
Precisely so, sorry: `None` is the default value meaning "no callback assigned". Because any routine that doesn't explicitly return a value in Python implicitly returns `None`, this also means that calling a function with no explicit return and assigning that to the callback will disable that callback.

One option I possibly should have considered was a special non-callable value other than `None` (e.g. a singleton class instance), but I'm afraid now it's established there's little we can do without a fairly major break with backwards compatibility. That said, I still think `None` is the natural choice for "no callback" in Python (it's certainly used in plenty of other frameworks in a similar manner) and the user does learn a valuable lesson in fixing this bug (that parentheses imply execution, while the function name alone is just referencing).

In other words, I'm not 100% convinced this requires fixing in the first instance though I do acknowledge it's likely to be encountered by people new to programming and that is admittedly a key audience for this library (an alternative view point would be that Python should warn about functions without explicit returns, or differentiate between functions that return values and those that don't ... but then that would smack horribly of VB's procedure/function distinction ;)

bennuttall, 2016-01-07:
Ah, that's a shame.

You're right, `None` is definitely the natural choice for "no callback". Makes perfect sense for un-setting a callback. Anything else would be a bit odd. Shame though - would have been good to catch that common error! But of course, a useful lesson!

closed 2016-01-07 by bennuttall

carlosperate, 2016-01-08:
Why can we convert `when_pressed` to a property and check the value assigned when set? Would that be a "not pythonic" thing to do?

bennuttall, 2016-01-08:
It is a property! See [input_devices.py#L133](https://github.com/RPi-Distro/python-gpiozero/blob/master/gpiozero/input_devices.py#L133)

The problem is, you should be able to unset the `when_pressed` event by setting it to `None`:

```python
button.when_pressed = None
```

However, if the user accidentally calls the function when passing it to `when_pressed`, it silently passes `None`. While this may surprise the user, it's expected behaviour. Observe:

```python
>>> def hello():
...:    return "Hello"

>>> a = hello()
>>> print(a)
Hello
```

```python
>>> def hello():
...:    print("Hello")

>>> a = hello()
Hello
>>> a
None
```

So effectively you are unsetting the event by accident.

waveform80, 2016-01-08:
Indeed, when execution "falls off the end" of a function without hitting a `return` statement, Python implicitly causes the function to `return None`. In other words, Python has no concept of a function that returns no value; a function *always* returns a value and if it doesn't do so explicitly, it will implicitly return the `None` value.

Now, given that `when_pressed` is indeed a property, we still need some value to indicate when the desired action is to "do nothing". I'd argue that the natural choice is the value `None` (indeed, this is what many event-driven frameworks use in such cases). However, it is true that I could have selected something different. I could have made the property accept callables only (i.e. throw an error if anything like `None` is assigned), and then used a callable which does nothing as the default value, e.g.:

```python
def do_nothing():
    pass

button.when_pressed = do_nothing
```

However, that would require that anyone wishing to unset the property firstly know about this "do_nothing" callable, import it from somewhere, and then use it (or define their own, of course).

Unfortunately it also leads to a certain amount of inefficiency. We can't know (without disassembling the provided callable) that it actually does nothing. As a result, we'd have to permanently install event handlers for everything that could possibly have them; at the moment we can use `None` as an indicator that we don't need to bother listening for certain events (we don't actually take advantage of this because we haven't needed to yet, but the option exists and this is one reason frameworks prefer non-callables for such defaults).

Notwithstanding the above, the backwards compatibility argument still trumps everything else. To change now would break all code that uses `None` to disable an event handler (this is a particularly common use case, especially when debugging at the command line). Sorry!

bennuttall, 2016-01-08:
Yeah I think we can agree there's nothing to be done. Thanks!

carlosperate, 2016-01-08:
Thanks for the extensive answers, I didn't realise `None` was an expected value to remove the callable. [gpiozero/input_devices.py#L145](https://github.com/RPi-Distro/python-gpiozero/blob/59ba7154c5918745ac894ea03503667d6473c760/gpiozero/input_devices.py#L145)

lurch, 2019-02-06:
Hmmm, just reading through this again (wow, 3 years later!) and a thought occurred to me...
Given that the default value of `.when_pressed` is `None`, could we emit a warning in the property-setter if the property was _previously_ `None`, and we're again setting it to `None`?
It might help people who get caught out by the `button.when_pressed = hello()` example, and I can't think of any legitimate scenarios where you'd deliberately set a callback to a `None` value if it was `None` already?

bennuttall, 2019-02-06:
Ooh that's a good idea

bennuttall, 2019-02-06:
Implemented in PR #726