gpiozero/gpiozero #140: Cannot install gpiozero due to missing dependency

SweetGale, 2016-01-07:
Hi! I've been trying to install the python-gpiozero package on Raspbian but am told that it cannot be installed due to one of the dependencies being unavailable. Anyone who can help me solve this? I've run `apt-get update` and made sure I'm on the latest firmware (in case that helps). Other than that I'm out of ideas.

Here's the output from apt-get (in English for your convenience):

```
pi@raspberrypi ~ $ LANG=en_GB.UTF-8 sudo apt-get install python-gpiozero
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 python-gpiozero : Depends: python-spidev but it is not installable
E: Unable to correct problems, you have held broken packages.
```
```
pi@raspberrypi ~ $ LANG=en_GB.UTF-8 sudo apt-get install python-spidev
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Package python-spidev is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'python-spidev' has no installation candidate
```

Output from apt-cache seems to indicate that python-spidev is a virtual package and that the functionality needs to be provided by another package. This is sadly going far beyond my knowledge of APT and I don't even know what type of information to look for at this point.

```
pi@raspberrypi ~ $ LANG=en_GB.UTF-8 apt-cache depends python-gpiozero
python-gpiozero
  Depends: python
  Depends: python
  Depends: python-rpi.gpio
  Depends: <python-spidev>
```

waveform80, 2016-01-07:
Ahh, would you perhaps be using Raspbian wheezy? It looks like python-spidev didn't get back-ported there - here's the output from `apt-cache depends python-gpiozero` on my jessie box:

```
 pi@raspberrypi ~ $ lsb_release -a
No LSB modules are available.
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 8.0 (jessie)
Release:	8.0
Codename:	jessie
 pi@raspberrypi ~ $ apt-cache depends python-gpiozero
python-gpiozero
  Depends: python
  Depends: python
  Depends: python-rpi.gpio
  Depends: python-spidev
```

Whereas on my wheezy box I see the same situation as yours:

```
dave@gonzo ~ $ lsb_release -a
No LSB modules are available.
Distributor ID: Debian
Description:    Debian GNU/Linux 7.8 (wheezy)
Release:        7.8
Codename:       wheezy
dave@gonzo ~ $ apt-cache depends python-gpiozero
python-gpiozero
  Depends: python
  Depends: python
  Depends: python-rpi.gpio
  Depends: <python-spidev>
```

You could download the python-spidev deb from the jessie repositories, install it manually with "sudo dpkg -i ..." and then install python-gpiozero but there's enough improvements in jessie that I'd seriously consider just making a new jessie SD card and going from there (don't try upgrading wheezy to jessie - it's too painful and risky - much better to start fresh on another card).

SweetGale, 2016-01-07:
Yep, there you have it, still running wheezy.

bennuttall, 2016-01-07:
Just mentioned this to @XECDesign who runs our Raspbian repo

XECDesign, 2016-01-07:
Copied the spidev packages across from jessie to wheezy, but unfortunately I can't test if this fixes the issue or introduces new ones.

waveform80, 2016-01-09:
Looks like spidev is now available in wheezy so you should be able to install with a straight forward "apt-get install" (run `sudo apt-get update` first to make sure your package list is up to date).

waveform80, 2016-01-09:
And for the sake of book keeping, this is related to #132

bennuttall, 2016-01-09:
Has anyone tested the install on Wheezy yet? Just want to check all's well before closing.

waveform80, 2016-01-09:
python-gpiozero works on my wheezy dev box; unfortunately python3-gpiozero doesn't because it depends on python3-spidev which, although now available on wheezy, unfortunately has been built on jessie so it includes the requirement python3 (>= 3.4~) which can't be met on wheezy (this is one of the reasons my dev boxes are stuck on wheezy, so I can build backwards compatible packages).

SweetGale, 2016-01-09:
The packages install successfully, yet for some reason Python cannot find the _spidev_ module on my box.

```
pi@raspberrypi ~ $ python -c 'import gpiozero'
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/usr/lib/python2.7/dist-packages/gpiozero/__init__.py", line 13, in <module>
    from .input_devices import (
  File "/usr/lib/python2.7/dist-packages/gpiozero/input_devices.py", line 15, in <module>
    from spidev import SpiDev
ImportError: No module named spidev
```

:confused:

waveform80, 2016-01-09:
Strange - that appears to work on my wheezy box ... let me see what's going on. Oh damn it, looks like I'd done `pip install --user spidev` at some point in the past (probably when I was building the gpiozero packages otherwise I doubt they'd have built successfully given it wasn't deb packaged at the time) which explains why it works for me. I'll dig into why the spidev package isn't working in a mo but I bet it'll be another artefact of it having been built on jessie...

waveform80, 2016-01-09:
Eurgh ... this was hard to track down. Apparently it's down to the [multi-arch support](https://wiki.debian.org/Python/MultiArch) added in Debian Jessie. Basically, packages containing "external" (i.e. C compiled) modules change slightly in multi-arch so that instead of being called, for example, `spidev.so`, they're now called `spidev.arm-linux-gnueabihf.so` (in other words, they're named module.*platform*.so instead of module.so).

As the multi-arch page linked above makes clear, the old naming system works too ("... extensions are looked up in a multiarch specific filename, before looking in the default filename") so this is a forwards compatible scheme (extensions compiled with the old naming convention will still work). Hence, if the spidev package were built under wheezy (without multi-arch support) it would use the old naming convention but would still work on jessie. However, as the package has been built on jessie it requires a version of python with the multi-arch patches to find it in the path correctly.

So, there's a couple of things that can be done here:

1. If we're serious about supporting gpiozero on wheezy as well as jessie we need to persuade the spidev maintainer to rebuild the packages for wheezy.

2. Failing that (or as a stop-gap measure) I can make spidev an optional import so that we only attempt to import it if someone attempts to use `AnalogInputDevice` (which is the only class currently using it). I must admit I'm tempted to do this anyway as this removes the last mandatory dependency and means gpiozero will work "out of the box" with PyPy (albeit not every class) once the pins stuff is merged.

Thoughts?

bennuttall, 2016-01-09:
Is 1 not something @XECDesign can do on the Raspbian side?

2 sounds ok

waveform80, 2016-01-09:
I believe so - you just need a wheezy box set up with all the necessary stuff for building packages, but there may be some minor alterations to make to the package sources (I've tried to have a look at the package sources but for some reason apt-get source isn't working for me - probably something wrong with my old dev-box setup - I'll dig into it in a mo).

waveform80, 2016-01-09:
Okay, source package was just missing from wheezy; it's present on Jessie. I think the only thing that needs altering *might* be to downgrade the Standards-Version in the debian/control file (I've got 3.9.3 in my picamera control file, but it's pretty old now so it's equally possible I'm just out of date), but everything else looks like it should build out of the box on wheezy.

bennuttall, 2016-02-10:
What's the status of this? Anything we need to do?

waveform80, 2016-02-10:
I've just run an upgrade on my wheezy build box and python-spidev hasn't been re-built yet. So it's your call really: we can keep pressing for a rebuild of the package on wheezy or I can make spidev an optional dependency (in which case wheezy users will either get an error whenever they try and use SPI based stuff like MCP3008 or they can run the `pip install --user spidev` line I mentioned above to work around it).

waveform80, 2016-02-10:
(we could mention that command in the import error for spidev as a prompt for wheezy users?)

bennuttall, 2016-02-10:
Let's do that. Optional dependency. More trouble than it's worth. If people want to use an advanced component on an old distro release they can figure out how to install what they need with pip, or ideally move to Jessie!

However, that would require the latest gpiozero being uploaded to wheezy, wouldn't it? Did 1.1?

Which reminds me.. #185

lurch, 2016-02-11:
Optional dependency sounds good to me. But if the `python-spidev` package that @XECDesign added to the Raspbian repos for Wheezy isn't working propely, should it be removed again from the Wheezy package lists?

waveform80, 2016-02-11:
1.1's on wheezy and jessie; I would assume 1.2 will be on both too (I'm keeping my package build boxes on wheezy specifically for compatibility)

keithellis74, 2016-02-14:
I'm getting similar issues on wheezy

    kellis@birdcam ~ $ sudo apt-get install python3-gpiozero
    Reading package lists... Done
    Building dependency tree       
    Reading state information... Done
    Some packages could not be installed. This may mean that you have
    requested an impossible situation or if you are using the unstable
    distribution that some required packages have not yet been created
    or been moved out of Incoming.
    The following information may help to resolve the situation:
    
    The following packages have unmet dependencies:
     python3-gpiozero : Depends: python3-spidev but it is not going to be installed
    E: Unable to correct problems, you have held broken packages.
    kellis@birdcam ~ $ 

python3-spidev is required, but

    kellis@birdcam ~ $ sudo apt-get install python3-spidev
    Reading package lists... Done
    Building dependency tree       
    Reading state information... Done
    Some packages could not be installed. This may mean that you have
    requested an impossible situation or if you are using the unstable
    distribution that some required packages have not yet been created
    or been moved out of Incoming.
    The following information may help to resolve the situation:
    
    The following packages have unmet dependencies:
     python3-spidev : Depends: python3 (>= 3.4~) but 3.2.3-6 is to be installed
    E: Unable to correct problems, you have held broken packages.
    kellis@birdcam ~ $

bennuttall, 2016-02-14:
Try:

```bash
sudo apt-get install python3-dev python3-pip
sudo pip3 install gpiozero
```

keithellis74, 2016-02-14:
This worked for me:
    
    git clone https://github.com/RPi-Distro/python-gpiozero.git
    sudo python3 setup.py install

closed 2016-04-01 by waveform80

sweenig, 2017-05-01:
I can't get this to work on Wheezy. I did
`sudo apt-get update
sudo apt-get upgrade
sudo apt-get install python-spidev
sudo pip install --user spidev`

I still get:
`pi@raspberry ~ $ python -c "from gpiozero import RGBLED"
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/usr/local/lib/python2.7/dist-packages/gpiozero/__init__.py", line 7, in <module>
    from .input_devices import (
  File "/usr/local/lib/python2.7/dist-packages/gpiozero/input_devices.py", line 11, in <module>
    from spidev import SpiDev
ImportError: No module named spidev
pi@raspberry ~ $`

Unfortunately, i don't have time at the moment to rebuild this box on jessie. I've tried in the past with not enough time and for some reason my stuff isn't working, so I don't want to try on this one with not enough time to troubleshoot my stuff to get it working.

bennuttall, 2017-05-01:
*Note: I've edited your comment to use three backticks around your commands to format it correctly*

You are installing spidev twice, once with apt, once with pip. I don't think that's what's causing your problem but you should only be trying one way or the other.

Do the install commands work successfully? Does `python -c "import spidev"` work at all?

Usually, `ImportError: No module named X` means you haven't got X installed. I'd follow up on why spidev isn't installing properly.

sweenig, 2017-05-01:
That's really odd. I hit the button to put that text into a code block. Looks like the behavior of the button is different depending on the order. If you put the text in first, select it, then click the button, it surrounds inside pairs of three ticks. If you click the code button first, it just puts in two ticks. If you paste your text in between, it doesn't make it code. 

I knew i was installing spidev twice, but it didn't work after either one and i was just summing up everything I had tried. I figured either one should have solved the problem, but i couldn't get it work.

```
pi@raspberry ~ $ python -c "import spidev"
Traceback (most recent call last):
  File "<string>", line 1, in <module>
ImportError: No module named spidev
```
This is what i get when trying to install spidev via pip:
```pi@raspberry ~ $ sudo pip install spidev
Requirement already satisfied (use --upgrade to upgrade): spidev in /usr/lib/python2.7/dist-packages
Cleaning up...```

This is what i get if i install it from apt-get:
```pi@raspberry ~ $ sudo apt-get install python-spidev
Reading package lists... Done
Building dependency tree
Reading state information... Done
python-spidev is already the newest version.
0 upgraded, 0 newly installed, 0 to remove and 5 not upgraded.
```

sweenig, 2017-05-01:
So, i went to install it manually (not really knowing how to do that, but willing to bumble my way through it). This seems to be the source: https://pypi.python.org/pypi/spidev

```
pi@raspberry ~/spidev-3.2 $ sudo python setup.py build
running build
running build_ext
building 'spidev' extension
creating build
creating build/temp.linux-armv6l-2.7
gcc -pthread -fno-strict-aliasing -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fPIC -I/usr/include/python2.7 -c spidev_module.c -o build/temp.linux-armv6l-2.7/spidev_module.o
spidev_module.c:20:20: fatal error: Python.h: No such file or directory
compilation terminated.
error: command 'gcc' failed with exit status 1
```
Seems it fails on the gcc (a c compiler?). Is this in reference to the comment above about the name change so that it now includes the platform name?

lurch, 2017-05-03:
>  If you put the text in first, select it, then click the button, it surrounds inside pairs of three ticks. If you click the code button first, it just puts in two ticks.

Single ticks are for inline code-blocks `like this`, whereas three ticks are for multi-line code-blocks
```
like
  this
example
```

> Seems it fails on the gcc

A missing `Python.h` error usually means that you don't have the `python-dev` package installed, which you can install with `apt-get`.

bennuttall, 2017-05-03:
Have you installed the latest gpiozero with pip? I believe the wheezy apt package will not be up-to-date, so you'll need to install it with pip.

What is your actual issue? You don't need spidev to use gpiozero. If you're using SPI devices like MCP3008 then it's recommended you do, but more recent versions of gpiozero allow you to fall back to a software implementation. In v1.0, spidev was a dependency, but we later removed the dependency as it's possible to use it without. If you actually need spidev, then you can install it with apt or pip, and as @lurch said you'll need python-dev/python3-dev to build it.

sweenig, 2017-05-04:
@lurch, you were right that python-dev was not installed. However, it doesn't seem to have helped. After `sudo apt-get install python-dev`, i get the following:
```
pi@raspberry ~ $ python -c "from gpiozero import RGBLED"
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/usr/local/lib/python2.7/dist-packages/gpiozero/__init__.py", line 7, in <module>
    from .input_devices import (
  File "/usr/local/lib/python2.7/dist-packages/gpiozero/input_devices.py", line 11, in <module>
    from spidev import SpiDev
ImportError: No module named spidev
```
I did not install gpiozero with pip. I actually didn't realize that i could as i was just following the main installation instructions on gpiozero.readthedocs.io, which say to install via apt-get. 
```
pi@raspberry ~ $ sudo pip install gpiozero
Requirement already satisfied (use --upgrade to upgrade): gpiozero in /usr/local/lib/python2.7/dist-packages
Requirement already satisfied (use --upgrade to upgrade): RPi.GPIO in /usr/lib/python2.7/dist-packages (from gpiozero)
Requirement already satisfied (use --upgrade to upgrade): w1thermsensor in /usr/local/lib/python2.7/dist-packages (from gpiozero)
Requirement already satisfied (use --upgrade to upgrade): spidev in /usr/lib/python2.7/dist-packages (from gpiozero)
Cleaning up...
```

My actual issue is that i couldn't create an RGBLED object. I wasn't (and am still not planning) on interfacing with anything via SPI (no ADCs or anything, just an RGBLED).
Here's what happens when i try to use pip to install spidev:
```
pi@raspberry ~ $ sudo pip install spidev
Requirement already satisfied (use --upgrade to upgrade): spidev in /usr/lib/python2.7/dist-packages
Cleaning up...
```

Given all the messages about using --upgrade to upgrade, i figured i'd muddle my way through that thinking maybe i had an old version. So i did a `sudo pip install spidev --upgrade`. 

...
...
...
...
Guess what now works!
```
pi@raspberry ~ $ sudo pip install spidev --upgrade
Downloading/unpacking spidev from https://pypi.python.org/packages/36/83/73748b6e1819b57d8e1df8090200195cdae33aaa22a49a91ded16785eedd/spidev-3.2.tar.gz#md5=f601676f1bb48b9aa3b3897f95216365
  Downloading spidev-3.2.tar.gz
  Running setup.py egg_info for package spidev

Installing collected packages: spidev
  Found existing installation: spidev 3.0
    Uninstalling spidev:
      Successfully uninstalled spidev
  Running setup.py install for spidev
    building 'spidev' extension
    gcc -pthread -fno-strict-aliasing -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fPIC -I/usr/include/python2.7 -c spidev_module.c -o build/temp.linux-armv6l-2.7/spidev_module.o
    gcc -pthread -shared -Wl,-O1 -Wl,-Bsymbolic-functions -Wl,-z,relro build/temp.linux-armv6l-2.7/spidev_module.o -o build/lib.linux-armv6l-2.7/spidev.so

Successfully installed spidev
Cleaning up...
pi@raspberry ~ $ python -c "from gpiozero import RGBLED"
```

I'll try using it in actual code, but the fact that it imports is a huge success. I learned something new: update things through pip as well as apt-get.

bennuttall, 2017-05-04:
This:

```
pi@raspberry ~ $ sudo pip install gpiozero
Requirement already satisfied (use --upgrade to upgrade): gpiozero in /usr/local/lib/python2.7/dist-packages
Requirement already satisfied (use --upgrade to upgrade): RPi.GPIO in /usr/lib/python2.7/dist-packages (from gpiozero)
Requirement already satisfied (use --upgrade to upgrade): w1thermsensor in /usr/local/lib/python2.7/dist-packages (from gpiozero)
Requirement already satisfied (use --upgrade to upgrade): spidev in /usr/lib/python2.7/dist-packages (from gpiozero)
Cleaning up...
```

indicates that you're on gpiozero ~v1.0, which depends on all those packages. As of gpiozero v1.2 it had no dependencies - which means you can do SPI without spidev. You need to upgrade gpiozero:

```
sudo pip install gpiozero --upgrade
```

lurch, 2017-05-04:
Glad to hear you persevered and finally got it working :-D
But I'm afraid the moral of the story is that Wheezy is now quite outdated and no longer supported.

PR #239 (2016-04-01): Fix #140, fix #69, fix #185
This PR adds a software SPI implementation. Firstly this removes the absolute necessity for spidev (#140), which also means when it's not present things still work (effectively fixes #185), and also enables any four pins to be used for SPI devices (which don't require the hardware implementation).

The software implementation is simplistic but still supports clock polarity and phase, select-high, and variable bits per word. However it doesn't allow precise speeds to be implemented because it just wibbles the clock as fast as it can (which being pure Python isn't act [... cut]
commit 759a6a58 (2016-04-01): Fix #140, fix #69, fix #185

This PR adds a software SPI implementation. Firstly this removes the
absolute necessity for spidev (#140), which also means when it's not
present things still work (effectively fixes #185), and also enables any
four pins to be used for SPI devices (which don't require the hardware
implementation).

The software implementation is simplistic but still supports clock
polarity and phase, select-high, and variable bits per word. However it
doesn't allow precise speeds to be implemented because it just wibbles
the clock as fast as it can (which being pure Python isn't ac [... cut]
commit a7b7fc8d (2016-04-01): Merge pull request #239 from waveform80/energenie

Fix #140, fix #69, fix #185